ArduPilot/ardupilot #20037: Mav proxy not connecting when using Airsim (Windows 10 + CygWin)

LightCannon, 2022-02-09:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**

I am facing a strange issue. When I run this (from ardupilot/ArduCopter folder):
…/Tools/autotest/sim_vehicle.py --console --map

everything runs well. But when trying to do the same but with **airsim**:
…/Tools/autotest/sim_vehicle.py -v ArduCopter -f airsim-copter --console --map

I get this
Connect tcp:127.0.0.1:5760 source_system=255
Loaded module console
Loaded module map
Log Directory:
Telemetry log: mav.tlog
Waiting for heartbeat from tcp:127.0.0.1:5760
MAV>

**Version**
latest master branch

**Platform**
[  ] All
[  ] AntennaTracker
**[  ]Copter**
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
SITL copter

**Hardware type**
SITL (using Airsim)

**Logs**
Not sure if this is the relevant log. Here is **ArduCopter.log** at /tmp/
Setting SIM_SPEEDUP=1.000000
Starting SITL Airsim type 1
Bind SITL sensor input at 127.0.0.1:9003
AirSim control interface set to 127.0.0.1:9002
Starting sketch 'ArduCopter'
Starting SITL input
Using Irlock at port : 9005
bind port 5760 for 0
Serial port 0 on TCP port 5760
Waiting for connection ....
Connection on serial port 5760
Loaded defaults from /cygdrive/f/CE/ArduPilot/ardupilot_base/Tools/autotest/default_params/copter.parm,/cygdrive/f/CE/ArduPilot/ardupilot_base/Tools/autotest/default_params/airsim-quadX.parm
bind port 5762 for 2
Serial port 2 on TCP port 5762
bind port 5763 for 3
Serial port 3 on TCP port 5763
MKFIFO failed with File exists
MKFIFO failed with File exists
Home: -35.363262 149.165237 alt=584.000000m hdg=353.000000
No sensor message received in last 1s, error - File exists, resending servos
No sensor message received in last 1s, error - File exists, resending servos
No sensor message received in last 1s, error - File exists, resending servos
No sensor message received in last 1s, error - File exists, resending servos
No sensor message received in last 1s, error - File exists, resending servos
No sensor message received in last 1s, error - File exists, resending servos
No sensor message received in last 1s, error - File exists, resending servos

DEV123458, 2023-09-20:
I am having the same issue.. Have anybody found solution?

iagupov, 2023-10-13:
The same for me, no solution found for the moment

eziosoft, 2023-11-01:
same for me

schlesg, 2023-11-10:
This seems to be happening only with new versions of Ardupilot. It works properly with version 4.3.3.