ArduPilot/ardupilot #20054: Tools: Plane's Landing-Drift auto test is flapping

rmackay9, 2022-02-11:


wp0351034, 2022-02-11:
您的邮件已经收到，尽快给您回复。北京堂仁翔科技有限公司联系人：王鹏  电话：135 2058 1985 ***@***.***网址：www.trxchn.comQQ:191896104地址：北京市海淀区花园北路14号环星大厦A座8层888室

peterbarker, 2022-02-11:
@samuelctabor FYI - we'd like to get this test fixed.  Ping me if you'd like some help looking at failing logs.

rmackay9, 2022-02-11:
It's just a guess but it may depend upon the state of the test run before it.

samuelctabor, 2022-02-15:
That's fine @rmackay9 .

It doesn't seem to be leftover state, the test resets SITL at the start and I ran it a few times locally without seeing the issue.

@peterbarker not sure how I can proceed without getting logs from the server

samuelctabor, 2022-02-22:
@rmackay9  @peterbarker  reminder . . . I could use the help getting this test enabled again.

rmackay9, 2022-03-24:
@samuelctabor @peterbarker this is fixed now right?

tridge, 2022-04-06:
seems to be fixed

closed 2022-04-06 by tridge