ArduPilot/ardupilot #20081: Rover SITL: crashes if vehicle turns too quickly

rmackay9, 2022-02-14:
I don't yet know if it is a Rover code issue or a simulator issue but the simulator crashes (or freezes) if the vehicle attempts a very high speed corner.  This can be reproduced by doing the following

- cd Rover
- sim_vehicle.py --map --console
- arm throttle (to arm in Manual mode)
- rc 3 2000 (100% throttle)
- rc 1 1000 (100% left steering)

[image]

xianglunkai, 2022-02-14:
This PR always exists in Rover simulation. I checked the code, and consider that it is produces by calculation of simulation model,including the integral divergence caused by lack of kinematic and dynamical constraints. Maybe should provide KT model and dynamical water simulation calculations for ships
  

 

 ------------------ 原始邮件 ------------------

 主题: [ArduPilot/ardupilot] Rover SITL: crashes if vehicle turns too quickly (Issue #20081)

IamPete1, 2022-02-14:
same as https://github.com/ArduPilot/ardupilot/issues/14469

Its not a rover issue, any vehicle will crash if you spin it fast enough. Just the rover sim is about the only way to get up to that speed.

rmackay9, 2022-02-15:
OK thanks, closing this one then.

closed 2022-02-15 by rmackay9