ArduPilot/ardupilot #20082: Arming Check for "GPS 3 fix obtained"

xfacta, 2022-02-14:
## Feature request
Arming check for "Home position can be set" or more technically "GPS 3D fix obtained" before the aircraft can be armed in any mode. It should be on by default since new users will expect the home position to be set by default or they wouldn't be allowed to fly. They will be expecting RTL to work even if they were able to arm and fly in Stabilise mode (for example)
A lot of new users wont be using a ground station.

For example in copter you can usually arm and fly in Stabilise or AltHold any time, without a 3D fix, unless you have a Fence set. In fact  documentation recommends to use these two modes for testing and tuning. Many people will not realise Home position hasnt been set.
Setting a Fence gives the desired action (cant arm and fly until Home can be set) but it is more difficult to find out about this feature and much less obvious than Arming Checks.
The reason for this new arming check to be on by default is: that would give a higher level of safety and confidence that RTL would work. Experienced users would know if they **dont** need a home position set (or are at least aware of the issues) and could easily find and disable this arming check (or update their Fence settings).

https://discuss.ardupilot.org/t/arming-check-how-to-use-a-bitmask-parameter/81801/5

**Is your feature request related to a problem? Please describe.**
There is currently no mandatory 3D fix requirement, depending on the flight mode selected.
It is difficult for a new user to equate Fence parameters with Arming checks, and may even assume that the "GPS" arming check means they should have a good 3D fix before being allowed to arm.
The current documentation needs updating to clarify arming, RTL and when a home position can or can't be set. I will attempt to update the documentation online for the existing situation.

**Describe the solution you'd like**
Arming check for "GPS 3D fix obtained" indicating that Home could be set when you arm. Arming would NOT be allowed until this condition is satisfied. Similar effect to trying to arm in Loiter mode without a 3D fix, or having a Fence set.

**Describe alternatives you've considered**
The alternative would be to make the Fence enabled by default, but that will be harder for new users to understand and find since the logical place to start looking for arming failures is in the Arming checks.

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
Unsure if this applies to more than just Copter, I suspect it could apply equally to Plane but not so much to Rover and Sub.

rmackay9, 2022-02-14:
The standard workaround we've recommended up until  now is to enable the fence.  This is a reasonable request though.

Quadzilla805, 2022-08-22:
"GPS 3D fix obtained" before the aircraft can be armed in any mode." Builders need a mode that we can arm 100 % without a 3D lock when on the workbench.  We also fly our crafts to test without GPS lock in Stabilize.
Moving an aircraft out doors during a build has it own sets of issue and could be dangerous.

xfacta, 2022-08-23:
Yes fair enough.
I usually "highly" recommend the fence setting because most users would arm
and launch then wonder why the aircraft is heading away at full speed.
For experienced people it would be easy enough to turn off the Fence
setting, rather than making inexperienced users go looking for something
they dont even know they should use.

peterbarker, 2024-07-31:
Closed by https://github.com/ArduPilot/ardupilot/pull/27367 - we now have a flight options bit to get the desired behaviour.

closed 2024-07-31 by peterbarker