ArduPilot/ardupilot #20092: Unknown WDG failures

Hwurzburg, 2022-02-15:
Report and logs from user on continuous WDG with T-3 and zero other data on recent master using KakuteF7
**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine
logs:
https://drive.google.com/file/d/1zhpmFaUxtocT8gXUGX8_Qo7hGyTNB_0A/view
https://drive.google.com/file/d/1minBTjC0KUiFwvQZRLpDv0-XCtfROkM2/view

rmackay9, 2022-04-29:
@Hwurzburg,

Do you think this is resolved by this PR https://github.com/ArduPilot/ardupilot/pull/20580?

I think I heard that it is rare enough that we can't reproduce it easily and so it is difficult to know if we've resolved it or not.  Maybe we should assume we've fixed it and re-open if it occurs again on a version which has the above PR included (e.g. 4.2.0-rc2 or higher)

Hwurzburg, 2022-04-29:
yep...never saw it again, but havent been using the the kakutef7 much...closing

closed 2022-04-29 by Hwurzburg