ArduPilot/ardupilot #20114: AP_Scripting: ahrs::get_location() is broken

rmackay9, 2022-02-18:
Lua scripts will fail with the syntax error shown below if they attempt to execute ahrs:get_location()

`attempt to call a nil value (method 'get_location')`

The problem is resolved if we remove this line from bindings.desc

`singleton AP_AHRS method get_location alias get_position`

rmackay9, 2022-02-18:
My guess is that it is early enough in the use of Lua that we can go with the new name and not cause much difficulty to users.

peterbarker, 2022-02-25:
This is fixed in master.

At the DevCall we mused on the possibility of being able to properly deprecate functions in LUA so people have an opportunity to migrate their scripts before we kill entry points.

rmackay9, 2022-02-25:
Ok, let's close this because the immediate issue is resolved at least.

closed 2022-02-25 by rmackay9