ArduPilot/ardupilot #20135: IMU position offset calibration/parameters

ES-Alexander, 2022-02-21:
## Feature request

**Is your feature request related to a problem? Please describe.**
Vehicle designs can't always incorporate the flight controller (or an external IMU) at/close to the desired centre of rotation. When that's not possible the flight controller fights against itself when performing correct/desired rotations because the IMU measures translations that it incorrectly tries to counteract.

**Describe the solution you'd like**
We already have multiple calibration options to correct for the orientation of the flight controller (e.g. specify orientation, calibrate accelerometer+gyroscope+compass), so some form of coordinate transform is already occurring. I want that transform to be expanded to include a position offset between the flight controller's actual location and the desired control/rotation centre.

At some point that offset may be automatically determinable, but for most use cases it should be sufficient for the user to be able to just enter values manually (from physical or CAD-based measurements), so this request is solely for the ability to enter an IMU position offset and have it form part of the internal coordinate transform.

**Describe alternatives you've considered**
The current approach to dealing with this is [to avoid it](https://ardupilot.org/copter/docs/common-mounting-the-flight-controller.html#:~:text=It%20should%20be%20placed%20close,and%20level%20with%20the%20motors.) by just placing the flight controller as close as possible to the vehicle's natural centre of rotation and using/assuming that as the control centre. Given ArduPilot is (as I understand it) about empowering users and developers to make and control their own vehicle designs, this seems like an unnecessary design restriction.

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[x] Submarine

**Additional context**
This is discussed in additional depth in [this forum comment](https://discuss.bluerobotics.com/t/auv-vehicle-gliding-forward-while-its-not-supposed-to-do/11581/5). It's relevant to all vehicle types, but particularly so for subs, where motors are defined with 6DoF thrust factors (relative to the control centre) and enclosure requirements can make it particularly challenging to position the flight controller near the _desired_ control centre.

This feature also opens the door to future dynamic control centre positioning, where perhaps something like a "cylinder inspection flight mode" sets the control centre to outside, in front of the vehicle, which would significantly simplify the process of making a smooth circular rotation around a pole/structure while consistently looking in towards it (would just require commanding the vehicle to turn slowly).

rmackay9, 2022-02-21:
I think we already have this in the form of the INS_POS_X/Y/Z parameters.  [Some details are on this wiki page](https://ardupilot.org/copter/docs/common-sensor-offset-compensation.html).  Sorry if I've misunderstood.

ES-Alexander, 2022-02-21:
That was an astonishingly quick response, and it does seem that you're correct (not sure how I didn't know about that already - the parameters have clearly been around for a while).

Do you have experience with how the coordinate transform is done? I'm not sure why it's "only partial" (as described in the page you linked to), unless the accelerometer is perhaps much faster to respond than the gyroscope or something. I would expect the transform to happen _before_ the data gets to the EKF, but that apparently isn't the case.

> Maybe we should just add a link to the sensors offset page (linked above) from [this wiki page](https://ardupilot.org/copter/docs/common-mounting-the-flight-controller.html).

I expect that would be helpful :-)

IamPete1, 2022-02-21:
Calculation is here:

https://github.com/ArduPilot/ardupilot/blob/bcf3f223758f623bfb640a9d7fe10619eadc149f/libraries/AP_NavEKF3/AP_NavEKF3_core.cpp#L888-L898

ES-Alexander, 2022-02-22:
> Calculation is here:
> ...

Appreciate the link. Unfortunately reading through that snippet, and the surrounding code and comments, confirms that the acceleration data isn't corrected but doesn't seem to clarify why. Perhaps it's a sensors relative frequency issue, or just wanting to avoid coupling the accelerometer and gyroscope noise or something.

While I'd like to know the actual reason, if the designers of that component/algorithm are unlikely to clarify then this issue can be closed, given the proposed feature at least mostly exists already.