ArduPilot/ardupilot #20140: Rover: enabling bendy ruler leads to a 1sec delay at each waypoint

rmackay9, 2022-02-22:
If Rover uses Bendy Ruler then whenever the vehicle passes a waypoint and a new destination is loaded, the vehicle stops until bendy ruler determines that there is a safe path to the next destination.  This takes about 1 second and leads to the vehicle stopping (or at least slowing down) momentarily at each destination.

A solution might be to modify bendy ruler to accept not only the current destination but also the next destination.  Bendy ruler is named "bendy ruler" because it sometimes uses a two part ruler so perhaps when the vehicle is very close to the destination, it could use the 2nd portion of the ruler to check from the current to the next destination.

xianglunkai, 2022-02-22:
I think whether we can increase the obstacle avoidance planning time from 1Hz to 10Hz, so that the deceleration will not be so obvious, and improve the effect of BR obstacle avoidance. In addition, if we consider shoreline obstacle avoidance and shoal obstacle avoidance, it should also be useful to improve the frequency of obstacle avoidance planning.

rmackay9, 2022-02-22:
@xianglunkai, that's a very good idea.

timtuxworth, 2022-02-22:
Could it be configurable?

rmackay9, 2022-02-22:
@timtuxworth, we could make it configurable if necessary but I suspect we can just make it 10hz and it'll just work for everyone.  I think maybe we slowed it down from 10hz to 1hz because we were trying to reduce the "hesitancy problem" (where the vehicle keeps changing it's mind about which path around the obstacle to take) but @rishabsingh3003 fixed this now.

timtuxworth, 2022-02-22:
> @timtuxworth, we could make it configurable if necessary but I suspect we can just make it 10hz and it'll just work for everyone. I think maybe we slowed it down from 10hz to 1hz because we were [...]

I'm thinking about how this might be used in a plane. Is 10hz going to be enough with a vehicle moving at 20,30,40 m/s?

rmackay9, 2022-02-23:
@timtuxworth, I think 10hz will probably be enough because very fast moving vehicles also tend to be very far from obstacles so they also have more time to react.  BendyRuler isn't enabled for plane yet though in any case.  Only Copter and Rover.

timtuxworth, 2022-02-23:
I have ambitions of enabling BendyRuler in plane. My questions are about things I need to be be aware of to make that happen.

manavgandhi17, 2022-03-08:
Adding that the vehicle also comes to a stop at the intermittent "waypoint" that is calculated for avoiding the fence(s). From what I read above, updating the algorithm to 10Hz may fix that issue as well?

xianglunkai, 2022-03-08:
I think your problem may be caused by AC_Avoid library. May be you should ensure that OA_MARGIN >> FENCE_MARGIN.

manavgandhi17, 2022-04-01:
> @timtuxworth, I think 10hz will probably be enough because very fast moving vehicles also tend to be very far from obstacles so they also have more time to react. BendyRuler isn't enabled for plane yet though in any case. Only Copter and Rover.

Can you direct me as to what files to edit to test this?

manavgandhi17, 2022-04-10:
Just pinging to see if there are any other places that might be causing this delay?

closed 2023-12-27 by rmackay9