ArduPilot/ardupilot #20158: Rover/Boat: autonomous docking

rmackay9, 2022-02-24:
We should add an autonomous docking feature (perhaps a new flight mode) to the Rover/Boat software that autonomously maneuvers the vehicle to within cm of a visual target.  This mode is similar in many ways to [Copter's precision landing](https://ardupilot.org/copter/docs/precision-landing-with-irlock.html) feature so we could probably re-use the AC_PrecLand library to estimate where the target is.  The controls (target speed and turn rate) will probably be quite different than Copter's though.

P.S. I thought we already had an issue for this but I can't seem to find it.

xianglunkai, 2022-02-24:
This is essentially a problem of changing one attitude to another, where the attitude includes position and heading. In ROS the navigation_experimental function pack set, there is a question about pose_ base_ controller function package, which can solve the attitude control problem of this mobile robot. This may help you.

i-m-karanagarwal, 2022-04-04:
I would like to work on this issue for my GSOC project.

lucifer79gg, 2022-04-15:
I would love to work on it for my GSoC. Would request you to please kindly assign it to me.

rmackay9, 2022-04-15:
Hi @lucifer79gg,

The best way is to make an application through https://summerofcode.withgoogle.com to the ArduPilot project.  txs!

sablin39, 2023-01-31:
I am wondering if this can still be a GSoC project since it appears on the idea list of 2023.

rmackay9, 2023-01-31:
Hi @sablin39,

No, I think we've completed this project so it won't be on the GSoC 2023 list.  We actually haven't decided on the project list for 2023 yet.

I think I'll close this to-do now..