ArduPilot/ardupilot #20160: Copter: very short SCurves take too long to complete on landing

IamPete1, 2022-02-24:
When running a mission with target landing location SCurves take a long time to complete resulting in a long hover. 

This is the test mission:

```
1	0	3	16	0	0	0	0	-35.3628498	149.1659045	10	1
2	0	3	16	0	0	0	0	-35.3628498	149.1659045	5	1
3	0	3	21	0	0	0	1	-35.3628498	149.1659045	0	1
```

A waypoint at 10m, a second waypoint at 5m then a land, all with the same lat/lon.

This shows the issue with standard SILT defaults, but with lower accel/jerk limits it can take as much tens of seconds. 

[image]

We see the landing start with the message `Mission: 3 Land` it then takes 3 seconds to get the `LANDING_GEAR_DEPLOY` event showing that landing descent has started. 

In this time the target altitude moves by 2cm,
[image]

The target NE position does not change at all. 

SCurves are calculating that it takes 3 seconds to move 2cm.

The reason this holds us up is the scurve section must be finished for the destination to be marked as reached. 

https://github.com/ArduPilot/ardupilot/blob/020262bfe7755951ec655e9257084c6920b0f642/libraries/AC_WPNav/AC_WPNav.cpp#L519-L533

Once `s_finished` is true the waypoint radius check completed instantly, as the vehicle has been within the way-point raduis the whole time. 

One fix would be to stop checking the if the scurve has finished and rely on only the waypoint radius. 

```
    // check if we've reached the waypoint
    if (!_flags.reached_destination) {
        // "fast" waypoints are complete once the intermediate point reaches the destination
        if (s_finished && _flags.fast_waypoint) {
            _flags.reached_destination = true;
        } else {
            // regular waypoints also require the copter to be within the waypoint radius
            const Vector3f dist_to_dest = curr_pos - _destination;
            if (dist_to_dest.length_squared() <= sq(_wp_radius_cm)) {
                _flags.reached_destination = true;
            }
        }
    }
```

I suspect we would see the same issue with normal waypoints if they were placed extremely close, but its easy say that that is not recommend, this landing case forces the same situation. 

This [calculator ](https://www.trajectorygenerator.com/ojet-online/?xs0=0&xt0=0.02&xs1=0&xt1=0&xs2=0&xt2=0&xp0=1000&xn0=-1000&xp1=1.5&xp2=1&xp3=1&mode=0&tvec=nt&ntdt=10000#start) suggests the movement should take 0.86 seconds. 

In anycase, I think the key question here is if we should wait for the SCurve to complete if within the waypoint radius. 

[00000007.zip](https://github.com/ArduPilot/ardupilot/files/8132822/00000007.zip)

Tested with https://github.com/ArduPilot/ardupilot/pull/20099 but does not fix.

lthall, 2022-02-24:
I am aware of this problem. It is caused by the Jerk time being fixed. So for these small segments we keep the Jerk time fixed and reduce the Jerk. 

A single leg being made up of 8 of these segments means that there is a minimum time that a leg can take to complete.

To fix this problem I need to recalculate all the S-Curve equations based on a max snap calculation where we reduce both the Jerk and Jerk time together by the same factor. This will make these little 2cm legs go much faster.

I don't think this is a enough problem to justify adding complexity and potentially bugs to the mission. The simple fix is don't use ridiculously short waypoint segments.
For this to be a problem we need:
- Very short waypoint legs
- Low Jerk and high acceleration

Yes, this is annoying but it is a silly corner case. 
Yes it needs to be fixed but it will take me week of work to do it and a lot of testing after that. 
There is no simple fix other than removing short waypoints from the mission either manually at the user level or automatically in the autopilot.

IamPete1, 2022-05-17:
Closed by https://github.com/ArduPilot/ardupilot/pull/20204

closed 2022-05-17 by IamPete1