ArduPilot/ardupilot #20166: Copter: arming allowed within fence margin

tridge, 2022-02-24:
With a polyfence (and possibly tin can?) if you are within fence margin of the fence you can arm but the copter will immediately tilt hard to get out of the margin region. This led to tipovers on 3 quads in testing till we worked out the cause. 
example log:
http://uav.tridgell.net/tmp/copter-fence-arm-margin.bin

peterbarker, 2024-07-31:
I've just verified that this is still an issue in master.

An appropriate prearm check is probably to call `AC_Avoid::adjust_velocity` - if it would adjust the velocity then refuse arming.

We would only make that call if the results of the call would be used - that's tied up with state in `AC_Loiter`.  Not all modes you might arm in use AC_Loiter....

peterbarker, 2024-07-31:
tridge thought this might work on DevCallEU

rmackay9, 2024-07-31:
Yes, the vehicle is moving because avoidance's backup feature is enabled.  So if we re-word the check so it is a check of whether avoidance would cause the vehicle to move immediately upon arming or not then this approach seems OK to me.