ArduPilot/ardupilot #20196: lightwareserial SF11C wrong baudrate handling

tridge, 2022-02-28:
A SF11C set as 57600 instead of the correct 115200 gave garbage data of 161 meters.
We need more robust checks that the data it is getting is valid

WickedShell, 2022-02-28:
Do we have a contact to request a checksum protocol from? It obviously won't help with any shipped units, and I'm definitely in favor of better error range checking, but this is exactly the kind of thing a checksum is designed to help avoid for the future.

rmackay9, 2022-03-01:
We have good contacts with Lightware and we could ask for their help.  They have various protocols available but maybe we are using an old one.

I expect that if the RNGFNDx_MAX_CM is set that it will catch this made distance and it won't be consumed by either the EKF or (copter's) terrain following.  I'm not saying there isn't a problem but we do have 2ndary checks that the distance is valid.

WickedShell, 2022-03-01:
The max range might help, but it's very plausible to accidentally parse
something inside the range limit, which would escape it. It narrows the
window, but by no means eliminates it the way a better protocol would.

tridge, 2022-03-01:
I think we should apply some stricter checks even without the checksum, for example rejecting the legacy format if it isn't a sequence of digits, full stop and linefeel/cr
for the binary protocol I think we need to use the "frame gap" method, like we use for RC protocols. So reset the parsing when there is a time gap of more than XX ms, and check we get exactly the number of bytes we expect