ArduPilot/ardupilot #20218: Potential unexpected operation of sprayers, parachutes, etc.

Hwurzburg, 2022-03-04:
With the removal of BRD_PWM_COUNT, there is a potential for unexpected operation of parachute/cargo releases, sprayers, etc. since many hwdef had pre-defined GPIO outputs using it and user may be counting on it to be a GPIO to perform the correct operation, ie be active low to NOT activate a release....but these automatically will become PWM outputs upon upgrade of firmware and could lead to unexpected operation of devices immediately upon firmware reboot....pre-arm checks to validate that assigned pins are GPIOs will not have any preventative effect on these issues...

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

Hwurzburg, 2022-03-04:
perhaps something can be done with SERVO_GPIO_MASK or changing defaults for SERVOx_FUNCTION to -1?

WickedShell, 2022-03-04:
> changing defaults for SERVOx_FUNCTION to -1?

This would make the upgrades for relays (such as in camera triggers) probably a lot less painful.

IamPete1, 2022-03-04:
I don't think `SERVOx_FUNCTION` 0 will actually output PWM, so it should not be triggering stuff by going high. Would have to check. 

Defaulting to -1 does mean GPIO will still work....

IamPete1, 2022-03-05:
I just double checked, we do not output PWM if set to 0. So setting to -1 does not make any difference in that regard. Of course if the user has something setup as a output that was previously covered by PWM mask then it will get PWM still. Changing the default won't help. I think at the time we just decided this was too complicated to deal with correctly.

Hwurzburg, 2022-03-05:
but does it force the output to active low? if not, then a release could occur...if it does, with either 0 , or -1 (assuming the release is active high, which is the only way it can be setup to prevent releases as the FC goes thru the bootloader (since it forces outputs active low) then there is no issue...

IamPete1, 2022-03-05:
It certainty goes low, I'm not sure what the exact state is. I'm not really sure what we can do about it even if it does go high. The only path I see to fixing this in all cases is to force reset all servo outputs to -1 and having the user set them up again.