ArduPilot/ardupilot #20224: Feature Request: Add version number to firmware file name

booo, 2022-03-05:
## Feature request

**Is your feature request related to a problem? Please describe.**

At the moment the version is not encoded in the filename of the firmware file. If you e.g. download different firmware versions it's hard to track which file is which version. You have to rename the files manually to keep track of the versions.

**Describe the solution you'd like**

Add the version number to the firmware file. Example: "arduplane-4.1.7.apj"

**Describe alternatives you've considered**

There is a version file present in the download directory. If you download the whole directory version information stays with the firmware file.

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

hendjoshsr71, 2022-03-05:
Hi Philipp,

In case you didn't know. You can open the .apj files with a text editor and
find a bunch of information: version, git hash, free flash, and more.

Of course not as convenient as your suggestion.

IamPete1, 2022-03-07:
Build target would also be useful. I have lots of files all over called `arduplane.apj`.  I'm too lazy to actually open the file and look.....

rmackay9, 2022-05-16:
This idea was discussed on the dev call and we agreed that it was a good idea and the final file name should include the vehicle name +  board name + version (e.g. "arducopter-CubeOrange-bdshot-4.3.0-DEV").  Note that all files that appear in the folder (e.g. .bin. etc) should use the new name (not just the .apj file)

We probably need to:

- modify the build scripts to include a new argument that allows the caller to specify the name of the output file (?)
- enhance the manifest generator to use the new names
- test various downstream system including MP's SITL work after the change

hendjoshsr71, 2022-05-16:
Additional notes from the devcall (since Randy beat me to publish :) ):

We would just do this for 4.3 latest and beta builds to shake out any tooling issues.
And rely on the manifest generation to shake out any issues with users of these files such as Mission Planner, QGC, etc.

timtuxworth, 2022-05-17:
I suggest the version number go at the front. Remember - these will be sorted alphabetically. Once there are a few of these in a folder (especially on an SD card) - what order would you want to see them in? Likely you are searching for a specifically version number and the name might get truncated in some cases. Also it might be problematic for some operating systems to use "." embedded in the name. The . can be replaced with - or _. Perhaps it would make more sense to be:

version-board-vehicle 
e.g. 4-3-0-CubeOrange-ArduCopter 
or 4_3_0_CubeOrange-ArduCopter

rmackay9, 2022-05-17:
I agree with @timtuxworth that we should avoid "." in the name except for the one before the "apj" or "hex".  I'm less sure about the suggestion on the order of the words though.  I think the order should be frame-board-version..

timtuxworth, 2022-05-17:
Actually we should avoid "-" too. Because "PixHawk1-1M" as a board name has a "-" in it, it will make it difficult for scripts that are processing lists of files and parsing out details. Perhaps "_" for all separators?

hendjoshsr71, 2022-05-17:
From user perspective I can sort of understand [VERSION]_[BOARDNAME]_[VEHICLE].[apj/hex/blah] ordering.

How many users commercial or otherwise use a bunch of vehicle types? 
Is what is most important to most users the vehicle type or version?
The vehicle upfront is nicer for devs and follows our current firmware.ardupilot.org folder structures

In the end probably not that many switch back and forth once they make the switch.  

So..... I might still lean to [VEHICLE]_[BOARDNAME]_[VERSION].[apj/hex/blah]  and alphabetical sort should still work on the sd card either way. 

On the you only can display some many chars thing can you think a case for that today??

timtuxworth, 2022-05-17:
> On the you only can display some many chars thing can you think a case for that today??
Not sure if there is an ArduPilot specific example, but I found that when I was flashing OpenTX firmware the file picker only showed the first part of the filename, so I renamed the file with the version at the front so I could pick the right one.

timtuxworth, 2022-05-17:
> From user perspective I can sort of understand [VERSION]_[BOARDNAME]_[VEHICLE].[apj/hex/blah] ordering.
>
> How many users commercial or otherwise use a bunch of vehicle types? Is what is most [...]

I think that kid of supports my point @hendjoshsr71 - most people will be working with a particular vehicle at least, so having vehicle at the front of the name is just extra stuff they need to ignore. At very least I'd argue that vehicle should not be at the front for this reason. 

I also suggested this because when building waf puts the bin files in a folder named for the board, so (at least for developers) the board name is also redundant unless you are moving the files somewhere common - like an SD Card I guess, but would you really mix different targets on the same SD card?

timtuxworth, 2022-05-17:
Don't forget the original request was to include the version number in the filename. Adding the board/target came later. Would @booo want the version number first to the board first?

DavideLentini, 2022-05-17:
> Don't forget the original request was to include the version number in the filename. Adding the board/target came later. Would @booo or @DavideLentini want the version number first or the board first?

Hi , sorry for the late reply . I would prefer the board first . Example : arducopter_[board_name]_[version_number].apj
Thanks all