ArduPilot/ardupilot #20242: get_pilot_desired_lean_angles

HefnySco, 2022-03-07:
https://github.com/ArduPilot/ardupilot/blob/f2fe55b5dae643a53833324b474b9bdf2a10139d/ArduCopter/mode.cpp#L429

I noticed that in function **void Mode::get_pilot_desired_lean_angles(float &roll_out, float &pitch_out, float angle_max, float angle_limit) const**

**roll_out** is updated by this equation:
roll_out = (18000/M_PI) * atanf(cosf(pitch_out*(M_PI/18000))*tanf(roll_out*(M_PI/18000)));
while **pitch_out** is left without modification. I tried to comment this line and printf the result as shown in video.

https://youtu.be/IVF-Qkuw15Y

I could not find any difference.. I suspect this might be a redundant line.

The linde does not corrupt any data... it just does nothing...unless I miss something here.

IamPete1, 2022-03-07:
It will reduce roll output when at high pitch. I do tend to agree that we could just remove it...

HefnySco, 2022-03-07:
@IamPete1 I thought it was a strange line here, because either this line does nothing or there should be an equivalent line for pitch against roll.

I have submitted a [PR](https://github.com/ArduPilot/ardupilot/pull/20244) for it.

HefnySco, 2022-03-08:
I made a table with excel and changed pitch for different ranges of Roll and the difference between original output and the output after applying this formula is less than 0.00009 that seems negligible to me -not very sure if it actually is-. 
[image]

Could you explain why there is no a similar formula for pitch for quadcopter when roll is very high as pitch and roll have the same roles in quad copters ?

Thanks.

HefnySco, 2022-03-08:
> Your table looks wrong. I get 28 for 45 pitch 38 roll.
> 
> (180/pi) * atan(cos(45 * (pi/180)) * tan(38 * (pi/180)))
> 
> https://www.wolframalpha.com/input?i=%28180%2Fpi%29+*++atan%28cos%2845+*+%28pi%2F180%29%29+*+tan%2838+*+%28pi%2F180%29%29%29

True I missed that the input are in centi-degrees
[image]

HefnySco, 2022-03-08:
> I would guess its roll only because pitch is the "priority" axis. Your more likely to want to go forwards/backwards than left/right.

I believe your guess makes a lot of sense . but if true that means that quadcopter moving from side to side is not as stable as a quadcopter moving forward or backward.

closed 2022-03-15 by IamPete1