ArduPilot/ardupilot #20251: autotest broken on macOS

andyp1per, 2022-03-08:
It's been like this for several months on Sierra and Monterey

[image]

## Bug report

autotest never gets going

**Version**

master

**Platform**
[ X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
N/A

**Hardware type**
N/A

**Logs**

timtuxworth, 2022-03-08:
I added a comment in the PR. It's still not working for me.

andyp1per, 2022-03-08:
You also need this fix: https://github.com/ArduPilot/ardupilot/pull/20254

timtuxworth, 2022-03-08:
I'll try that. I also needed to install gdb. It's no longer included in Xcode tools. I got the Macports version, but now the dumpstack*.out file is giving me this:

```
[New Thread 0x1b03 of process 24494]
[New Thread 0x1c03 of process 24494]
[New Thread 0x1d03 of process 24494]
[New Thread 0x1f03 of process 24494]
[New Thread 0x2003 of process 24494]
Error calling thread_get_state for GP registers for thread 0x1b03

warning: Mach error at "/opt/local/var/macports/build/_opt_bblocal_var_buildworker_ports_build_ports_devel_gdb/gdb/work/gdb-11.2/gdb/i386-darwin-nat.c:132" in function "virtual void i386_darwin_nat_target::fetch_registers(struct regcache *, int)": (os/kern) invalid argument (0x4)

warning: unhandled dyld version (17)
0x000000030d8f0000 in ?? ()
#0  0x000000030d8f0000 in ?? ()
No symbol table info available.

Thread 5 (Thread 0x2003 of process 24494):
#0  0x000000030d8f0619 in ?? ()
No symbol table info available.
Backtrace stopped: Cannot access memory at address 0x0

Thread 4 (Thread 0x1f03 of process 24494):
#0  0x0000000000000000 in ?? ()
No symbol table info available.
#1  0x0000000000000000 in ?? ()
No symbol table info available.

Thread 3 (Thread 0x1d03 of process 24494):
#0  0x000000010ddd5000 in ?? ()
No symbol table info available.
#1  0x0000000000000000 in ?? ()
No symbol table info available.

Thread 2 (Thread 0x1c03 of process 24494):
#0  0x000000010ddd5000 in ?? ()
No symbol table info available.
#1  0x0000000000000000 in ?? ()
No symbol table info available.

Thread 1 (Thread 0x1b03 of process 24494):
#0  0x000000030d8f0000 in ?? ()
No symbol table info available.
A debugging session is active.

	Inferior 1 [process 24494] will be detached.

Quit anyway? (y or n) [answered Y; input not from terminal]
```

timtuxworth, 2022-03-08:
> You also need this fix: #20254

I tried that - it didn't help. SITL runs fine until I set the mode to Auto.

This happens armed or not. if I set the mode to Auto I get "link down" (has sitl

closed 2022-03-08 by peterbarker

andyp1per, 2022-03-08:
@timtuxworth don't use gdb on macOS - use lldb

timtuxworth, 2022-03-08:
> @timtuxworth don't use gdb on macOS - use lldb

I didn't do it - somewhere in sim_vehicle.py it's calling gdb. I agree with you lldb makes more sense.

timtuxworth, 2022-03-09:
> @timtuxworth don't use gdb on macOS - use lldb

How?

peterbarker, 2022-03-09:
> > @timtuxworth don't use gdb on macOS - use lldb
> 
> How @andyp1per ?

`--lldb` rather than `--gdb`

 -- advanced andyp1per substitute

timtuxworth, 2022-03-09:
Im not using --gdb. Im only passing --console, see above. So I dont understand why its runnung gdb.

andyp1per, 2022-03-09:
It does it by default, you have to explicitly tell it to use lldb instead