ArduPilot/ardupilot #20264: CUAV-x7 IMU2 gyro orientation incorrect?

rmackay9, 2022-03-09:
Below is a screen shot from a user log involving a crash of a Copter running V4.1.5 (stable) on a CUAV-x7 board.  The vehicle flipped upside down apparently because the AHRS incorrectly decided that the 2nd EKF core's attitude was best but it was actually completely incorrect. Why it made the incorrect decision is one question but more than this it appears (to me at least) that the 2nd IMU's gyro orientation is incorrect.  In particular it seems that it's Gyro Z value is the mirror image of the 1st IMU's GyroX value (so one of them is surely wrong).

[image]

Note that this board was running the -bdshot version of the firmware although I don't think this should matter.

I have the full log if required.

Vkd4U2NtVkZOVg, 2022-03-09:
Hello @rmackay9, I want to know which sensor did this X7 used? The X7 did replace some IMU sensors due to the global silicone shortage?

cuhome, 2022-03-11:
@Vkd4U2NtVkZOVg
[image]