ArduPilot/ardupilot #20277: Copter: Auto takeoff gets stuck if alt-above-home is positive and takeoff alt is terrain

rmackay9, 2022-03-10:
As reported by @Hasi123 here Auto mode (and presumably Guided as well) can get stuck on a takeoff command if an alt-above-terrain is used.  To reproduce the issue in SITL do this:

- Start SITL copter at the regular location
- create a short mission with a takeoff command followed by a Waypoint command where both have altitude frames of "AGL/Terrain

[image]

- module load message
- long MAV_CMD_DO_SET_HOME 0 0 0 0 -35.363 149.165 575  <-- to set a home altitude 
- param set AUTO_OPTIONS 3 <-- to allow arming and taking off in Auto
- auto
- arm throttle

You should see the Copter gets stuck and won't proceed to the waypoint command

rmackay9, 2022-03-10:
This is maybe fixed by https://github.com/ArduPilot/ardupilot/pull/20153

Hasi123, 2022-03-10:
I would like to add, that it only gets stuck if you start the mission (switch to AUTO) somewhere, where the terrain is lower than at home. So to reproduce:

- Start SITL in a hilly/mountainous environment
- Create and upload a mission starting with a takeoff command in AGL/terrain frame and a waypoint (doesn´t matter if its terrain/relative/absolute)
- Arm & takeoff in loiter/guided and fly downhill (somewhere, where the terrain is lower than at home)
- switch to AUTO and copter will get stuck after climbing to target altitude

rmackay9, 2022-03-11:
@Hasi123,

You probably saw my ping already but maybe you could confirm that this issue is resolved in master now?

Hasi123, 2022-03-12:
I'll only be able to test it next week.

Hasi123, 2022-03-14:
> @Hasi123,
> 
> You probably saw my ping already but maybe you could confirm that this issue is resolved in master now?

Just tested it in SITL, seems to be working now as intended.

rmackay9, 2022-03-14:
@Hasi123, great! thank you very much for finding this issue and helping us fix it!

closed 2022-03-14 by rmackay9