ArduPilot/ardupilot #20354: long spool time problem on quadplanes

tridge, 2022-03-22:
some ESC types need a long spool time to cope with variable time for motors to start (eg. APD pro ESCs)
A value for Q_M_SPOOL_TIME of 2s may be used. This applies to both spool up and spool down.
This gives a few problems:
 - switching to MANUAL mode to stop motors leaves the motors at the previous command PWM for half of the spool time
 - in forward transition if one motor happens to be high at the end of the transition (due to roll/pitch demands) it will remain at fixed high demand for half the spool time

Example:
[image]

lthall, 2022-03-22:
It looks like plane has stopped running the attitude control while the motors are still running. It looks like the simple fix is to command zero roll, pitch, yaw, thrust while you wait for the motors to spool down.

WickedShell, 2022-03-22:
The plane commanded a desired spool state of shut down the moment it went manual, but during that time the motors do the behavior above.

IamPete1, 2022-03-22:
I suspect there are several issues in this area where were not using the attitude control / motors libraries in the way they are expecting us to. Spool logic is certainly somewhere where we fall short. When we first arm for example we set the desired spool state and immediately start running attitude control. What we should be doing is staying relaxed until the actual spool state reaches unlimited.

I don't think we have any major hidden bugs, but this area does certainly cause twitches and results in vehicle behaving in unpredictable ways. It is quite a bit of work to bring plane's management of this stuff up to the same level as copter. 

Related to https://github.com/ArduPilot/ardupilot/pull/20255#issuecomment-1062029472 the problem there is that we just stop calling `motors->update()` on tailsitters and tiltrotors in some cases. I guess this log is a normal quad-plane so that is not the issue were seeing here?

WickedShell, 2022-03-22:
>  guess this log is a normal quad-plane so that is not the issue were seeing here?

Normal quadplane, those are the motor output channels plotted. The general expectation is that when in manual there are no quad motors, and the quadplane code just sets the desired spool state to shut down, which while it may not be sufficent, it's also very surprising to me at least that during the initial ramp down the motor outputs are actually frozen, when we are trying to get to stopping. An instant jump to throttle minimum then spooling would make more sense to me, but I may be missing a lot more.

tridge, 2022-03-22:
fixed in #20355

peterbarker, 2026-11-15:
Closed by PR #20355 (d82a306cc2, in 4.3+). QuadPlane::set_desired_spool_state() now zeroes the roll/pitch/yaw/throttle demands when SHUT_DOWN is requested

closed 2026-11-15 by peterbarker