ArduPilot/ardupilot #20362: CubeOrange: blheli passthrough cannot be used reliably with AUX1

rmackay9, 2022-03-23:
During DShot/BLHeli testing we discovered that CubeOrange's AUX1 servo output cannot be used for BLHeli passthrough (see wiki here).  It is not clear if the issue only affect BLHeli passthrough or if it also causes DShot or ESC Telemetry to not work either.

This is probably a hardware design issue to do with some protection circuitry on the servo outputs because the issue was not found on CubeBlack nor other autopilots from other manufacturers (in particular the CUAVv5).

joe-stroup, 2023-05-19:
I've just installed the Cube Orange BDSHOT firmware on two cubes - one connected to a T-Motor F55A Pro2 esc, and one connected to a Holybro Tekko32 esc.  The pass-through worked OK on Aux-1 for the T-Motor esc.  It still failed on the Holybro. 

These were two different cubes - running the same firmware - but there may have been minor differences in the parameter settings.

BTW - the ArduPilot wiki page that points to this open issue (#20362)  erroneously states that BDSHOT can't be used on Aux-1 because of this issue.  BDSHOT works fine on Aux-1 - just not passthrough.  I got good rpm data from motors on Aux 1-4.

[image]

[image]

[image]

Ryanf55, 2025-04-19:
Passthrough works fine for me with Lumenier Razor Pro F4 45A and CubeOrangePlus on AUX1 and AUX2

andyp1per, 2025-11-03:
There have been numerous passthrough fixes since this. I am going to close unless it can be proven this is still an issue.