ArduPilot/ardupilot #20366: Rover: add support COLREGs avoidance for dynamical avoidance?

xianglunkai, 2022-03-23:
As the simulation shows, bendyruler local avoidance is not compliance with COLREGs，and dynamical obstacle avoidance effect is poor.
Do we need to adjust the course and speed according to the relative speed, and add the planning of speed.

rmackay9, 2022-03-25:
@xianglunkai, txs very much for the suggestion.

For others reading here is a link (just the first one I found) for [COLREGs](https://www.imo.org/en/About/Conventions/Pages/COLREG.aspx) which describes how boats should avoid hitting each other.

xianglunkai, 2022-03-25:
Recently, I developed an obstacle avoidance algorithm based on speed obstacle method. This method complies with maritime rules on a wide sea surface, but it still needs to be improved in the face of crowded lakes.
By the way, do we need to use AIS as an other input for AP_Avoidance，so that boat at sea or on the river also be helpful!

closed 2022-03-31 by xianglunkai

bijuthaj, 2025-01-23:
Just curios...Is COLREG support is added for rover? @xianglunkai @rmackay9