ArduPilot/ardupilot #20376: Custom compass rotation broken

andyp1per, 2022-03-25:
On my copter I have the following custom rotation:

[image]

on 4.1 this works fine. On 4.2 I get an internal error:

[image]

**Version**
4.2beta2

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad

**Hardware type**
MatekH743-bdshot

**Logs**

IamPete1, 2022-03-25:
@andyp1per what is `COMPASS_ORIENT`?

The changes to custom rotations are not in 4.2 but If you had been on master and then back to 4.2 that would cause the issue your seeing here. The param conversion does not work on the downgrade. Fix is just to set `COMPASS_ORIENT` back to 100 manually.

andyp1per, 2022-03-25:
It didn't happen again, so I think you must be right