ArduPilot/ardupilot #20406: Rover: scurve navigation with bendyruler could not avoid obstacle and resume original route

xianglunkai, 2022-03-31:
[image]

xianglunkai, 2022-04-03:
I don't think scurve navigaiton has more advantages than L1 method in Boat. Besides, it has better turning performance. Moreover, the turning performance depends on very large deceleration. From my test, the L1 method is sufficient in terms of track mapping accuracy, obstacle avoidance effect and automatic control under high flow rate, at least for the current master branch.

rmackay9, 2022-04-04:
@xianglunkai, thanks very much for testing and reporting issues with the new SCurves on Rover.  We've still got time to fix any issues before 4.3 and I would very much like it if we can convince you to use SCurves because I really want to avoid two different versions.

I guess the picture above is not of master Rover?  I guess instead this is a modified version that gives up on the current waypoint and moves to the next one if it sees an obstacle?  I'm just trying to understand how I could reproduce the issue you're seeing and help resolve it.

xianglunkai, 2022-04-04:
Finally, whether we could design a trajectory input interface in guide mode for the navigation needs of ROS or other systems. Maybe I could give some code for the development of obstacle avoidance and planning system, but I still need your help.
Txs!

rmackay9, 2022-04-04:
@xianglunkai.

Re the issue of the Rover not returning to the original path, you've tired setting AVOID_OPTIONS = 0 of course?

rmackay9, 2022-04-05:
@xianglunkai,

OK, thanks.  Do you think the issue is new to 4.3?

xianglunkai, 2022-04-05:
@rmackay9 Yes,I am sure this pr is new for current master

xianglunkai, 2023-05-04:
@rmackay9 
I think L1-NAV coulde be used during obstacle avoidance and scurve be used after obstacle avoidance, which can solve two problems:

1. The problem of hesitation caused by severe deceleration of the curve during obstacle avoidance

2. The problem of avoiding obstacles and restoring the original route

Looking forward to discussion, if feasible, I would like to submit a fix.
thank you!