ArduPilot/ardupilot #20410: DataFlash binary logs are corrupted whenever transfer from the AutoPilot to computer through USB(with MissionPlanner) is cancelled

pere2504, 2022-03-31:
## Bug report

**Issue details**
DataFlash binary logs are corrupted whenever transfer from the AutoPilot to computer through USB(with MissionPlanner) is cancelled

Step to reproduce
1. Produce a lot of heavy log file (200 Hz logging rate, around 50-200 MB file size, around 30 logs)
2. Plug AutoPilot to computer with USB cable
3. Connect to AP through MAVLINK COM port
4. Under FlighData/DataFlash log/ --> Download DataFlash log via Mavlink
5. On the popup window select Download all logs
    * Progress bar does not show
    * However a message notice the log transfer has begun
    * Wait for about 5-10 min but nothing happen
6. Cancel transfer
7. Warning window prompting to tell transfer has not finished yet
8. Confirm cancel transfer
9. Redo steps 2 to 5
10. MissionPlanner now says there is no more log present on the AP

When plugging the SD card directly in the computer, we can see the file but they are corrupted. The name of every item are made of cryptic characters.

I am not sure whether it is an ArduCopter issue, a MissionPlanner issue or a MavLink issue. This whole issue seems weird to me since data should not be able to be corrupted in read mode.

**Version**
Firmware: 4.1.5
MissionPlanner: 1.3.76 Build 1.3.8029.15962
Computer: Microsoft Windows 10

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
OCTOQUAD X

**Hardware type**
CUAV X7 Pro

pere2504, 2022-03-31:
> Try increasing the log buffer size parameter.

Yeah I think it would help for futur flight. However, the fact remains that how is it possible to corrupt data in a copying process (read only context)? In the past I experienced other weird & random error when downloading log file. I wonder if something could be made to make the process more robust.

For now, I plan to always remove the SD card whenever I want to download critical data. I trust Windows or MacOS more than ArduPilot/MissionPlanner for this process.

timtuxworth, 2022-03-31:
Try QGroundControl. I find it works well and is much faster than Mission Planner.

amilcarlucas, 2022-03-31:
What makes you think that the files get corrupted by the copy process?
They are corrupted in the card. So I do not think that the reading process corrupts them. I think the writing process corrupts because of buffer underruns.

closed 2022-04-01 by pere2504