ArduPilot/ardupilot #20417: Copter: scripting's set_target_angle_and_climbrate's use_yaw_rate fields don't work

rmackay9, 2022-04-01:
Our lua script bindings allow a copter's angle (in euler angles) and climbrate (in m/s) to be controlled by a lua script.  The [binding](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Scripting/generator/description/bindings.desc#L223) also includes a couple of extra fields that should allow specifying the yaw rate as well.  I think the intention was that the if the yaw rate is provided that the yaw angle is ignored but in any case, it seems in the Copter implementation ([see Copter::set_target_angle_and_climbrate()](https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/Copter.cpp#L394)) we don't use these two fields at all.

khanasif786, 2022-04-01:
@rmackay9, using yaw_rate means ignoring yaw angle, but what about roll and pitch. Are they also to be ignored ?

rmackay9, 2022-04-02:
Hi @khanasif786,

I think if yaw rate is used then we should just ignore the yaw angle.  So roll and pitch angle should still be used. thanks!

khanasif786, 2022-04-02:
@rmackay9 . As i looked at the code internally it uses attitude target which doesn't support rates (see screenshot below). Please correct me if somewhere i am wrong.
[image]
.

khanasif786, 2022-04-03:
As mentioned by @lthall here #20435 . We should remove yaw rate completely. For that purpose i have made a seprate PR (link below).