ArduPilot/ardupilot #20438: Support for IRC Tramp protocol to control VTX parameters

Inv3nt0r1, 2022-04-03:
## Feature request

**Is your feature request related to a problem? Please describe.**
Not really. I want the VTX to run in Pit mode until I arm. When I am on the bench, the VTX gets really hot.

**Describe the solution you'd like**
Just like Smart Audio, I would like Ardupilot to support the IRC Tramp protocol to control VTX parameters like frequency and power level. An option to keep the VTX in pit mode until we arm.
I see more VTXes coming to support IRC tramp compared to Smart Audio.
Currently, betaflight and Inav support it. Being an Ardupilot pilot, I would like to have these features in my FPV drones having IRC Tramp VTX. 

**Describe alternatives you've considered**
NA

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
PR of IRC Tramp in betaflight: https://github.com/betaflight/betaflight/pull/2106

zxt2010, 2022-04-25:
I had the same problem with my panda vt5804.It only supports the IRC TRAMP protocol.

GaryWarren55, 2022-05-16:
I have just started using Ardupilot It has so many features. But Yes i agree this feature would solve my ability to control power levels on all my Matek 1.3 Vtxs a standard feature on Betaflight.

palladin82, 2022-05-29:
+up
Need to run in pit mode using IRC tramp when disarmed.

closed 2022-07-13 by andyp1per