ArduPilot/ardupilot #20444: Copter: Yaw-rate behavior in SET_ATTITUDE_TARGET changed in Guided

ajshank, 2022-04-04:
## Bug report
The new implementation of Mavlink's `SET_ATTITUDE_TARGET (82)` in Copter 4.2 has introduced a bug (or 'feature') -- it is no longer possible to set target angles for roll/pitch and only _angular rate_ for yaw for the attitude controller.

In 4.1, the `BODY_YAW_RATE_IGNORE` field was used to [select yaw vs yaw-rate](https://github.com/ArduPilot/ardupilot/blob/1002fd6e9728466d9eeaa338505e5b50d340d315/ArduCopter/mode_guided.cpp#L995). If `ATTITUDE_IGNORE` is false, and `BODY_YAW_RATE_IGNORE` is also false, 4.1's behavior was to choose yaw-rate over the yaw from attitude -- i.e., rate-ignores got priority. 

In 4.2, the implementation seems to use both. In [this commit](https://github.com/ArduPilot/ardupilot/commit/fd8c180f57132a7eda9a8afabf9edde38ad3f2c2#), the behavior is different (and somewhat strange). It seems like the rate controller is called for _all_ axes if input quaternion is invalid/zero, and not called if there's any valid attitude. This makes it impossible to run yaw-rate control without yaw. Furthermore, the [documentation here](https://ardupilot.org/dev/docs/copter-commands-in-guided-mode.html#copter-commands-in-guided-mode-set-attitude-target) may be outdated too.

We do a lot of high-level control which generates target roll/pitch angles, and target yaw-rates (sounds strange, but isn't unusual). It would be nice to retain support for independent yaw-rate control somehow.

### Feature
I think there's a broader question here -- what should be done if both attitude and rates are provided (and neither are set to 'ignore'). I would _not_ interpret rate to be the rate at which the provided attitude should be attained; but I suspect others might read it differently. @lthall, thoughts?

**Version**
4.2 // Copter 4.2.

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

rmackay9, 2022-04-04:
Txs for the report, I've added it to the [4.2 issues list](https://github.com/ArduPilot/ardupilot/issues/20192).

Just as a procedural matter normally we ask that issues are raised in the [4.2 forum ](https://discuss.ardupilot.org/c/arducopter/copter42/161)but it's ok this time.

lthall, 2022-04-04:
This is not a bug. 

SET_ATTITUDE_TARGET (82) does not support roll or pitch inputs at all. There is no way to specify euler angles, much less partially.

Attitude is completly specified by a single quaternion. Attitude is either used or ignored.

Rate inputs can be ignored but the only way to do that in reality is to assume they are set to zero when ignored.

The commit you mentioned fixes the very old, very incomplete and very broken interpretation of SET_ATTITUDE_TARGET (82).

>> We do a lot of high-level control which generates target roll/pitch angles, and target yaw-rates (sounds strange, but isn't unusual). It would be nice to retain support for independent yaw-rate control somehow
This is not supported by SET_ATTITUDE_TARGET (82). You should be using Quaternion and angular vectors as is clearly specified in SET_ATTITUDE_TARGET (82). You will find you get much more precise and responsive control.

If you want to suggest some other mavlink message that does support roll/pitch yaw rate inputs you are welcome to open a support request.

This is clearly not a bug.

rmackay9, 2022-04-04:
I've re-opened this because I think we may need to provide an alternative that allows control of the vehicle using roll and pitch angles and yaw rate.  So we can discuss what that is but we can't just consider this closed.

ajshank, 2022-04-05:
Thanks very much, @lthall and @rmackay9.
I can't agree that there's no way to specify Euler angles -- clearly, the specified quaternion can be converted to roll, pitch and yaw angles. The problem is with the rate targets (particularly in yaw), and how they should work with a quaternion target. I agree that Mavlink's poor definition of (82) leaves this quite open to interpretation.

> This is not supported by SET_ATTITUDE_TARGET (82). You should be using Quaternion and angular vectors as is clearly specified in SET_ATTITUDE_TARGET (82). You will find you get much more precise and responsive control.

It is "supported" in the sense that the message contains the field we need. We convert target angles to Quaternion, and have Copter ignore the yaw in it (because we are also sending target yaw-rate). I have used this from 3.5 to until 4.1, even before thrust-setpoint was supported. The implementation may have been hasty, but it provided a robust means to control the vehicle as a pilot would in `STABILIZE` mode.

Ideally, the Mavlink message should contain a mask for ignoring individual attitude axes (just like it does for individual rate axes). But, with the current `SET_ATTITUDE_TARGET (82)`, Copter can, say, _define_ three supported combinations:
- `STABILIZE/ALTHOLD`-style, where only roll/pitch are from quaternion, and yaw-rate is used.
- `ACRO/SPORT`-style, where only rates are used (require ATTITUDE_IGNORE).
- `STABILIZE`-style, but with full attitude from Quaternion (rates, if provided, used for feedforward).
   
I do understand the difficulties in making this explicit and unambiguous to all users though. I am happy to work towards this.

lthall, 2022-04-05:
>> I can't agree that there's no way to specify Euler angles -- clearly, the specified quaternion can be converted to roll, pitch and yaw angles.

Yes but it can't be converted to just roll and pitch angles, by definition. The Quaternion defines the complete attitude without any ambiguity.

>> I agree that Mavlink's poor definition of (82) leaves this quite open to interpretation.

This is actually a very good and VERY clear specification. The message clearly defines both the instantaneous attitude and/or angular velocity. This is exactly what is needed for fast precise attitude control.

>> It is "supported" in the sense that the message contains the field we need.

No, it isn't supported in even the most imaginative interpretation of the message. It is ridiculous to suggest that it is acceptable to butcher this specification as you describe here.

>>  I have used this from 3.5 to until 4.1, even before thrust-setpoint was supported. The implementation may have been hasty, but it provided a robust means to control the vehicle as a pilot would in STABILIZE mode.

And now the message is implemented properly and you can implement attitude control much more effectively and significantly more responsively. If you have been using this for that long it will not be a problem for you. In fact, if you have been using it for this long I am surprised you aren't dancing with joy that we finally fixed it properly.

This was discussed on the dev call again today and like each time before the conclusion was the same. This message has been implemented correctly and will not be changed. Using this message as it is right now you are able to implement any type of attitude control system you want with simple transformations at your end.

ajshank, 2022-04-05:
@lthall Thanks for offering your thoughts on this.
I posit that Mavlink only offers message definitions, and as such, cannot be clear on what to do with it. More generally, though, the fact that it doesn't offer masking for attitude axes makes it incomplete for several use-cases. In the same vein, `ATTITUDE_IGNORE` should make no sense for a message that is called `SET_ATTITUDE_TARGET`, but we happily do rate control when attitude is ignored. And therefore, even without a "most imaginative" interpretation, the message supported yaw-rate control, as Copter had [originally implemented](https://github.com/ArduPilot/ardupilot/blob/1002fd6e9728466d9eeaa338505e5b50d340d315/ArduCopter/GCS_Mavlink.cpp#L1090). And it surely wasn't ridiculous to have the `use_yaw_rate` bool this way.

> In fact, if you have been using it for this long I am surprised you aren't dancing with joy that we finally fixed it properly.

I haven't had the time, given that the "fix" actually breaks all of our current setup.
Adapting our outer-loops isn't a dramatic problem, but extensive retesting & re-verification across many platforms is a quite a bit. There's nothing to rejoice about, because the previous implementation wasn't even broken for a `STABILIZE`-like use-case.

I'm not calling the current implementation _incorrect_ - something must've been lost in communication here if it has come across so (I've referred to this as a new 'feature' above), and I apologize if that's the case. My point, in fact, is that it departs from previous behavior without retaining any means to obtain that same behavior, and without a full documentation of the change. I am also not against the current implementation -- I am merely stating the fact that it leaves no option available to mimic `STABILIZE` or `ALTHOLD`-like control, for whatever reasons one might need it. Perhaps there's yet another `DEV_OPTIONS` field that can be used for this.

> This message has been implemented correctly and will not be changed.

Thanks for such decisive clarification. Unfortunately, it isn't what one would expect from a massive open-source project.

lthall, 2022-04-05:
>> I haven't had the time, given that the "fix" actually breaks all of our current setup.
You are welcome to not upgrade to 4.2. That would minimise any need for "extensive retesting & re-verification" to check all the other changes in 4.2 I made. 

Or you could maintain your own patch on 4.2 that maintains the previous behaviour. Open source is great that way.

I am sure you will enjoy the additional flexibility and control the complete support of this message brings when you get around to updating your systems to 4.2.

ajshank, 2022-04-06:
Noting this for general reference (and for mine as well):
The "additional flexibility" you state is quite welcome, but if I'm not mistaken, this change will cause `Guided_NoGPS` to no longer work the same, and _may_ render it almost useless. Since `SET_ATTITUDE_TARGET` is the only message this mode works with, an external computer can no longer regulate the vehicle's heading -- yaw will always be controlled by the autopilot using the target quaternion. This would mean that (1) yaw control is now split from an outer-loop, and (2) the autopilot will now need yaw information sent to it from whatever positioning source was being used.

Here, (1) is definitely problematic for those who've worked with 4D control so far [3D pos + yaw]. And, with (2), might as well send positions too and move to purely `Guided` then.
Furthermore, [this doc on Guided_NoGPS](https://ardupilot.org/copter/docs/ac2_guidedmode.html#guided-nogps) will need to be updated.

ajshank, 2022-04-06:
Also, yes, I will need to implement some fix for this, because 4.2 is pretty great; all the work the devs and testers have put into it is phenomenal. Having previously maintained patches for this message (to enable `ThrustAsThrust` and `lift_max`), I would've preferred to avoid doing so yet again. I'll wait around a bit to see if there's an alternative resolution.
Thanks.

lthall, 2022-04-06:
Mate your statements just don't make sense. I am starting to think you don't understand how this works.

The only thing that Guided_NoGPS does is let you use the SET_ATTITUDE_TARGET without a gps lock. It is identical.

The only way this would "render it almost useless" is if SET_ATTITUDE_TARGET was useless. But it is more capable and flexible now than it was before. You can do more with it now than you could before, not less.

"an external computer can no longer regulate the vehicle's heading -- yaw will always be controlled by the autopilot using the target quaternion."

No, this is completly false. The quaternion comes from the message, from the companion computer, not the autopilot. The companion computer completly defines the attitude, including the heading.

I have a feeling you have done some crazy things in the way you have implemented your system to get around the previous limitations. I would suggest you take some time to review your approach and how it can be done better using the complete message in 4.2.

ajshank, 2022-04-07:
I'm sorry to say this, but I am getting the feeling that you aren't reading this carefully - which, really, is surreal.
> No, this is completly false. The quaternion comes from the message [..]

This is _exactly_ what I said - the quaternion target (from the message) will provide a yaw target for the autopilot; c.f. my statement above: "yaw will always be controlled by the autopilot using the target quaternion".
Previously, the yaw target from the quaternion to RPY conversion was discarded entirely if yaw-rate was provided. This meant that the companion computer didn't _have_ to define a heading for the autopilot -- maybe its estimate of yaw is different from Copter's, and it simply wishes to command a rotation rate.

> You can do more with it now than you could before, not less.

Please let us know if there's still a means to control yaw-rate only (without an explicit yaw-target).

lthall, 2022-04-07:
Ahh
>> an external computer can no longer regulate the vehicle's heading -- yaw will always be controlled by the autopilot using the target quaternion.

Here you say, the yaw CAN'T be controlled because the yaw IS being controlled by the quaternion. Even though the message is all coming from the same place. You had to quote only half the sentence to avoid the obvious. Lets not play games.

>> Please let us know if there's still a means to control yaw-rate only (without an explicit yaw-target).

This is not a support forum. Please post your questions here:
https://discuss.ardupilot.org/

lmark1, 2023-02-07:
Hello, 

I ran into the same issue as @ajshank when updating Copter firmware beyond 4.2.

Previously i was able to set roll and pitch targets in body frame along with yaw rate.  This enabled us to control the UAV with an external source of state estimation (Optitrack, SLAM, etc.). 

The only way this is possible now, as far as I can see, is by switching to full rate control, which might be a bit more risky to properly test.

ajshank, 2023-02-14:
Hi @lmark1 , 
Cross-posting here for reference:
> I agree that the current message structure doesn't support this feature, and hence, randomly changing it can be a disaster. My current solution is to add a `GUID_OPTIONS` field [to explicitly enable this feature](https://github.com/ajshank/ardupilot/commit/1c9c2fc89fd4ea319f6d8209dc4f605a51752fd1).

lthall, 2023-02-14:
>> The only way this is possible now, as far as I can see, is by switching to full rate control, which might be a bit more risky to properly test.

@Imark1 Can you explain why this can't be done with the existing quaternion based attitude message?

>>  Cross-posting here for reference:

This is the beauty of open source, if you don't like something you can just change it.

lmark1, 2023-02-15:
>  Can you explain why this can't be done with the existing quaternion based attitude message?

Hey @lthall, I was wrong. After making some changes to my position controller, I was able to produce the same functionality as prior to 4.2.

Changes are as follows:

- Initialize the desired yaw with the ATTITUDE_QUATERNION message, (a.k.a. mavros/imu/data if using ROS). Integrate the calculated yaw-rate target (based on some external yaw estimate and a given setpoint) and add it to the desired yaw value. This is similar to the control routine I was previously using: https://github.com/ArduPilot/ardupilot/blob/fe9461242e06aa8f74328ff1c7931990f5cd0687/libraries/AC_AttitudeControl/AC_AttitudeControl.cpp#L306
  (prior to 4.2 i was sending the desired yaw-rate straight to Ardupilot)
- Construct a quaternion from desired roll and pitch values (in the body frame) and the mentioned yaw (prior to 4.2 i was setting desired yaw=0)
- Set the type mask to ignore roll/pitch/yaw rates (prior to 4.2 i did not ignore yaw rates)

I guess this workaround makes more sense than what I was previously doing. 
Anyway, thank you for your time, patience and comments :+1:

lthall, 2023-02-15:
@lmark1 Nice!!!

I appreciate you going to the effort to try this!!

If you want a more responsive Yaw response you can provide just the yaw into the rate part of the message. You need to convert the yaw rate to the body frame rates though.

Feel free to DM me on any of the various ArduPilot discussion mediums if you would like to talk about this in more detail.

PeterQFR, 2023-06-10:
So has there been a change, I upgraded to 4.3.6 for use in no-gps mode and now my aircraft yaws immediately. 

I don't care about Yaw but only specify yaw rate, as well as pitch and roll in the body frame. as per @ajshank my yaw is not the autopilots yaw so whatever yaw is in that message should have been discarded. The issue is that with an external position estimate, the yaw can be arbitrary and not fixed to either north (if a compass is attached) or the power-on yaw if not using an autopilot with a compass. In the latter case the yaw transform is entirely arbitrary.

so @lmark1 your workaround is based on getting the autopilot's estimate for yaw and updating that?

PeterQFR, 2023-06-10:
Actually looking back through my code, maybe it was always broken for me. I had enabled body frame roll and pitch previously, here is the patch:

```
diff --git a/ArduCopter/mode_guided.cpp b/ArduCopter/mode_guided.cpp
index 0f4de592a5..d933843afe 100644
--- a/ArduCopter/mode_guided.cpp
+++ b/ArduCopter/mode_guided.cpp
@@ -635,7 +635,6 @@ void ModeGuided::set_angle(const Quaternion &attitude_quat, const Vector3f &ang_
     // convert quaternion to euler angles
     float roll_rad, pitch_rad, yaw_rad;
     attitude_quat.to_euler(roll_rad, pitch_rad, yaw_rad);
-
     // log target
     copter.Log_Write_Guided_Attitude_Target(guided_mode, roll_rad, pitch_rad, yaw_rad, ang_vel, guided_angle_state.thrust, guided_angle_state.climb_rate_cms * 0.01);
 }
@@ -1030,7 +1029,11 @@ void ModeGuided::angle_control_run()
     if (guided_angle_state.attitude_quat.is_zero()) {
         attitude_control->input_rate_bf_roll_pitch_yaw(ToDeg(guided_angle_state.ang_vel.x) * 100.0f, ToDeg(guided_angle_state.ang_vel.y) * 100.0f, ToDeg(guided_angle_state.ang_vel.z) * 100.0f);
     } else {
-        attitude_control->input_quaternion(guided_angle_state.attitude_quat, guided_angle_state.ang_vel);
+        float roll_rad, pitch_rad, yaw_rad;
+        guided_angle_state.attitude_quat.to_euler(roll_rad, pitch_rad, yaw_rad);
+        attitude_control->input_euler_angle_roll_pitch_euler_rate_yaw(ToDeg(roll_rad)*100.f, ToDeg(pitch_rad)*100.0f, ToDeg(guided_angle_state.ang_vel.z)*100.0f);
+        GCS_SEND_TEXT(MAV_SEVERITY_WARNING, "R %.2f, P  %.2f YR %.2f, yrcds %f \n", roll_rad, pitch_rad, guided_angle_state.ang_vel.z, guided_angle_state.yaw_rate_cds );
+        //attitude_control->input_quaternion(guided_angle_state.attitude_quat, guided_angle_state.ang_vel);
     }
 
     // call position controller
```

lthall, 2023-07-22:
I had a great chat with @Jaeyoung-Lim and he suggested that the desired behaviour could be achieved using the POSITION_TARGET_LOCAL_NED if we added support for MAV_FRAME_LOCAL_FRD frame.

https://mavlink.io/en/messages/common.html#POSITION_TARGET_LOCAL_NED
https://mavlink.io/en/messages/common.html#MAV_FRAME_LOCAL_FRD
https://mavlink.io/en/messages/common.html#POSITION_TARGET_TYPEMASK

We could support the input where only X and Y acceleration are not ignored. Then Alt_Hold could be controlled with rate and/or position. Yaw can be controlled with rate and/or angle.

This is a much more elegant and appropriate solution to this problem.

I am closing this PR in preference for https://github.com/ArduPilot/ardupilot/issues/24375

closed 2023-07-22 by lthall

IuliuNovac, 2024-01-04:
@lthall, Using a Quaternion for yaw control would have been ideal if the Attitude messages were delivered with almost no lag, and if there was no overshooting in the yaw 'set attitude' messages, especially at message frequencies around 100 Hz. However, in practice, things are quite different. We're seeing unstable yaw movements, with noticeable overshoot and jitteriness. If you are expecting to blame PID calibration for that, the odd part is that after auto-tune, pilot sees  no issues in yaw whatsoever when he is controlling from RC.

Yaw rate is necessary to compensate for these issues.

How to test it? Try rotating the drone by 10 degrees a second in a loop based on the current attitude from Mavlink + 10 degrees in radians, then use this value in set attitude. The behavior I saw is as if the drone is on crack.

lthall, 2024-01-05:
>> How to test it? Try rotating the drone by 10 degrees a second in a loop based on the current attitude from Mavlink + 10 degrees in radians, then use this value in set attitude. 

This sounds like you are doing something strange here. You need to provide a smoothly moving quaternion attitude OR a smoothly moving quaternion attitude and 3 axis rate (kinematically consistent). If you use the aircraft's current attitude as part of the calculation or only on axis of rate then you expect to get an aircraft that looks like it is on crack.

IuliuNovac, 2024-01-05:
> This sounds like you are doing something strange here. You need to provide a smoothly moving quaternion attitude OR a smoothly moving quaternion attitude and 3 axis rate (kinematically consistent). [...]

I think I'm not fully grasping how the set attitude is supposed to be utilized.

My system depends on visual data to gauge the discrepancy between the drone's desired attitudes and its actual attitudes. Where the desired attitude is the predicted attitudes to minimize the error from visual data. In order to provide the drone with set attitudes, it's crucial for me to know the attitudes my drone exhibited at the time when the measurements were taken. Given that disturbances like wind can significantly affect the system's capability to follow the specified attitude targets.

How can I be expected to provide smoothly moving quaternion attitude without the capability to receive feedback on current attitudes? If I am not aware of the specific attitudes the drone had during the error computation from the visual data, i can not able predict the desired attitudes and everything goes to shit.

IuliuNovac, 2024-01-05:
> This sounds like you are doing something strange here. You need to provide a smoothly moving quaternion attitude OR a smoothly moving quaternion attitude and 3 axis rate (kinematically consistent). [...]

I think I'm not fully grasping how the set attitude is supposed to be utilized.

My system depends on visual data to gauge the discrepancy between the drone's desired attitudes and its actual attitudes. In this context, the desired attitude refers to the predicted attitudes that minimize the error from visual data. This is crucial, especially given that disturbances like wind can significantly affect the system's capability to follow the specified attitude targets.

How can I be expected to provide smoothly moving quaternion attitude without the capability to receive feedback on current attitudes? If I am not aware of the specific attitudes the drone had during the error computation from the visual data, I cannot predict the desired attitudes, and everything goes to shit.

I would have preferred to use other commands, but I need to use GUIDED_NOGPS, making set attitudes my only option.

lthall, 2024-01-06:
Could you start a thread and repeat this question and the video question on the discus forums so we can talk there: https://discuss.ardupilot.org/