ArduPilot/ardupilot #20473: u-blox F9P firmware 1.30 is not working

tridge, 2022-04-06:
Our auto-config code is not working with the new 1.30 firmware from u-blox for the F9P
Several things don't work:
 - users lose all settings them they update to 1.30
 - the resulting settings don't allow ArduPilot to auto-detect the GPS
 - manual config at 230400 with UBX+NMEA+RTCMv3 allows detection, but auto-config doesn't complete
 - MON_VER message seems to not be supported, so firmware version can't be detected

khancyr, 2022-04-07:
Beside nmea 4.11 I didn't much important changes in the release note.

tridge, 2022-05-10:
i've re-tested this and 1.30 is working fine for me now on 3 F9P modules. I don't know why it failed for me in initial tests, but now everything does work. I've tested auto-config, RTK, moving baseline, multiple test flights. No issues found

closed 2022-05-10 by tridge