ArduPilot/ardupilot #20481: Ac4.2 Modalai voxl cam test issue

roque-canales, 2022-04-07:
Hello i’m trying to make the voxl cam work with ac4.2.

connexion is well established but i get issue:
[image]
Issue is that voxl tell that ap is upside down….
All is standard and viso orientation is set to forward
[image]
So i dont understand how to resolve this…
Can you help?
maybe @Tridge know how to do that if he can share with us ;)
also i confirm that i have good link between cube and modalai
 
[image]

rmackay9, 2022-04-19:
Hi @roque-canales, thanks for giving this a try.  The "can't initialise" message is not from ArduPilot but instead it comes from voxl.  Maybe the system is upside-down?  Seems unlikely and I'm sure confirmed the orientation yourself.

roque-canales, 2022-04-23:
Hello Randy, I also pushed issue on the modal ai gitlab:
https://gitlab.com/voxl-public/voxl-sdk/services/voxl-vision-px4/-/issues/6

I saw that tridge also tested voxl cam, it would be greacefull if he can help us on this testing.

Rgds

rmackay9, 2022-12-19:
Hi @roque-canales,

I think you've already noticed that we've fixed up the AP support for the ModalAI and created [a wiki page here](https://ardupilot.org/copter/docs/common-modalai-voxl.html).  The fixes are in Copter-4.3.2 which is available for beta testing now so if it's OK I will close this issue.  I will monitor for replies though so if you find problems I'm very happy to investigate.

closed 2022-12-19 by rmackay9