ArduPilot/ardupilot #20519: UART communication crash caused by armed operation

xianglunkai, 2022-04-13:
The current master version runs in the Linux environment has this problem.
When the armed is triggered, the uart communication(like /dev/ttymxc2) is faster and occasionally breaks down the normal serial communication

xianglunkai, 2022-04-13:
There was no such problem before version 4.2. 
I think it may be caused by modifying the serial port driver or scheduler code.
Txs!

rmackay9, 2022-04-13:
Txs for the report.  This is using standard AP of course on an RPI4 perhaps?  If there are any crash dumps or other instructions on how to reproduce this issue that would be greatly appreciated!

xianglunkai, 2022-04-14:
@rmackay9 Thank you. 
   I transplanted AP into an IMX cortext-A9 controller and developed a serial port servo driver with the underlying STM32 controller.
  Last year, when using version 4.1 program, everything worked well, but after the recent transplantation of version 4.3 program, the serial port communicating with STM32 will restart STM32 due to sudden failure to write data,When I trigger the unlocking operation. 
   by the way,I have observed that armed will cause the serial port to send more data in a short time, and other operations including reading and writing home points and setting modes will not cause this phenomenon.

xianglunkai, 2022-04-14:
@rmackay9 By the way, file write function return errno = 9,and return -1.

xianglunkai, 2022-04-15:
@rmackay9
 I found that the problem lies in the newly developed function Hal_ LOGGER_ FILE_ CONTENTS_ ENABLED = 1 in AP_Logger file!
  when Hal_ LOGGER_ FILE_ CONTENTS_ ENABLED  = 0, everything works well!!
  If you are free, can you help me review that code again.
  Thank you very much!

xianglunkai, 2022-04-15:
@rmackay9 This pr have been fixed now.
close it!

closed 2022-04-15 by xianglunkai

magicrub, 2022-04-16:
Can you share a link to your repo?

xianglunkai, 2022-04-16:
@magicrub https://github.com/ArduPilot/ardupilot/pull/20540
Txs，the pr has been fixed yesterday.