ArduPilot/ardupilot #20527: EKF: GPS Glitch can be misleading (because it's not always the GPS's fault)

rmackay9, 2022-04-13:
"GPS Glitch" is displayed to the user if the EKF fails to fuse a GPS sensor value (or perhaps several sensor values) because they are too far from it's current position estimate.  This inability to fuse though isn't always caused by the GPS glitching however so the message can be misleading it can cause the user to investigate their GPS quality when in fact the issue is that the vehicle's heading is off (e.g. a compass issue).

We should either make the message more vague (e.g. "position estimate glitch") or try and add some heuristics to better judge if the issue is actually GPS or is perhaps compass/heading sensor.

khanasif786, 2022-04-22:
@rmackay9 Sir, Please take a look.
[image]

rmackay9, 2022-04-23:
@khanasif786, very impressive!

closed 2022-05-18 by rmackay9

Hwurzburg, 2022-05-28:
@rmackay9 is this same message used if vibrations are affecting position also?

rmackay9, 2022-05-30:
@Hwurzburg,

Yes, the message could also be displayed if vibrations are pulling the position estimate away from the GPS.  We discussed this possibility just before merging this PR and we decided that if vibrations were the problem there would be lots of indications to the pilot of the problem so we figured we didn't need to add "vibration" to this message.

PR #20783 (2022-05-18): Copter:changed GPS glitch message
Fixes #20527