ArduPilot/ardupilot #20531: Found a bug related to RC3 parameter setting related to motor PWM setting

HanJIk, 2022-04-14:
When the RC3 min max value is changed from 1100, 1900 to 1000, 2000 and then operated, the changed range PWM value is not reflected and there is a bug that is still confirmed as 1100, 1900
1100 and 1900 values were entered as [hard coding](https://github.com/ArduPilot/ardupilot/blob/Sub-4.0/ArduSub/joystick.cpp#L113-L128) as the cause.

rmackay9, 2022-04-14:
I've added the Sub prefix and label because I think this is only related to Submarines but please correct me if I'm wrong.

ES-Alexander, 2022-05-16:
Noting that there's been a bunch of discussion about this in [this forum thread](https://discuss.bluerobotics.com/t/ardusub-servo-output-range-setting/11944).

In particular:

> we’ve since found that there are also other relevant parts of the code that use the same hardcoded limits (e.g. the * 400 in [this line](https://github.com/ArduPilot/ardupilot/blob/Sub-4.0/libraries/AP_Motors/AP_Motors6DOF.cpp#L229)), and finding and fixing all relevant sections may be challenging.
> ...
> Some of them won’t be essential to fixing the issue (although fixing some and leaving others could cause issues elsewhere). We haven’t gone looking for every case of it, or determined exactly what is required for enabling a larger output range. It’s possible it’s just the two things that have been linked to already, but also very possible there are other limitations elsewhere.

Williangalvani, 2022-05-16:
FYI, The RC3 issue was fixed in https://github.com/ArduPilot/ardupilot/pull/18576.

IamPete1, 2022-09-04:
Sounds like we can close.

closed 2022-09-04 by IamPete1