ArduPilot/ardupilot #20563: Rover: add support for mars rover style steering

rmackay9, 2022-04-18:
We should add support for the steering system used by the Mars rover.  [Here is a discussion on the Rover forums about this](https://discuss.ardupilot.org/t/esa-exomy-stem-mars-rover/83916).

srmainwaring, 2022-04-18:
@rmackay9 - if you're happy with a Lua script [I've previously posted something that will do this on discourse](https://discuss.ardupilot.org/t/enable-pivot-turn-for-custom-rover-frame/65161/2?u=rhys). It would benefit from an update, options to set parameters based on frame dimensions, etc.

It's also a nice model to have about if you're looking for examples to demo ROS integration, there's a ROS nav stack that goes along with it.

srmainwaring, 2023-01-10:
There is an example Gazebo model and script available in SITL/models: https://github.com/ArduPilot/SITL_Models/pull/95. The script has variables customised to the dimensions of the Sawppy rover. The following list outlines the tasks to turn it into a standalone applet for libraries/AP_Scripting/applets:

[ ] - add params for wheel coordinates / base dimensions.
[ ] - add params for servo allocation.
[ ] - add a slew rate / or stop when switching from standard to pivot style steering (at the point when the steering reverses as the instantaneous centre of curvature moves inside the wheel base radius).