ArduPilot/ardupilot #20566: Copter: GPS - on some board we have incorrect calculations of delta_time_ms

Sekilsgs2, 2022-04-18:
On some board (my Mambo F405 MK2, and AirbotF4 ) perhaps errors in design or due to a poorly calibrated oscillator or something else...
As a result, we have timimg drift between gps and mcu - but i'm think every board have some timing drift...
This moment is not correctly taken into account in the current implementation of the calculation - 
delta_time_ms - if i'm using gps_rate_ms == 200ms i'm have gpa.delta spikes (~340ms) every 20 seconds (see logs). This is make Unhealthy GPS signal warning every 20 sec - but at the same time, data from GPS arrives as expected without loss and in well-defined 200ms!.

To find what cause this i'm make some source code modifications in AP_GPS.cpp and in jitter_correction - 
```
        // delta will only be correct after parsing two messages
        timing[instance].delta_time_ms = tnow - timing[instance].last_message_time_ms;
        uint32_t lstm = timing[instance].last_message_time_ms;
        timing[instance].last_message_time_ms = tnow;
        // if GPS disabled for flight testing then don't update fix timing value
        if (state[instance].status >= GPS_OK_FIX_2D && !_force_disable_gps) {
            timing[instance].last_fix_time_ms = tnow;
        }
        static uint32_t ll_ms = 0;
        uint32_t nnow = AP_HAL::millis() - ll_ms;
        ll_ms = AP_HAL::millis();
        static uint32_t last_log = 0;
        if (((AP_HAL::millis() - last_log) > 1000) || nnow > 220 || timing[instance].delta_time_ms > 220) {
                GCS_SEND_TEXT(MAV_SEVERITY_INFO, "gg nn=%ld ll=%ld dtm=%ld tn=%ld lm=%ld", (uint32_t)nnow, ll_ms, (uint32_t)timing[instance].delta_time_ms, (uint32_t)tnow, lstm);
                last_log = AP_HAL::millis();
        }
```

This spikes make function  when - **min_sample_counter == convergence_loops**
**jitter_correction.correct_offboard_timestamp_usec** - 

```
    if (min_sample_counter == convergence_loops) {
        // we have the requested number of samples of the transport
        // lag for convergence. To account for long term clock drift
        // we set the diff we will use in future to this value
        link_offset_usec = min_sample_us;
        min_sample_counter = 0;
        GCS_SEND_TEXT(MAV_SEVERITY_INFO, "jitt %ld %ld %ld", (uint32_t)local_usec/1000, (uint32_t)offboard_usec /1000, (uint32_t)estimate_us/1000);  ///i'm add this log
    }
```

As we can see from logs - problem only in calculations of delta_ms but not GPS.
As a temporary solution - use gps_rate_ms - 190 - for me this fix any GPA.delta spikes and Unhealthy GPS signal warnings!

**Version**
4.0+

**Platform**
[ x ] Copter

**Logs**

```
18.04.2022 17:43:42 : gg nn=200 ll=175348 dtm=200 tn=175138 lm=174938
18.04.2022 17:43:41 : gg nn=202 ll=174346 dtm=200 tn=174138 lm=173938
18.04.2022 17:43:40 : gg nn=203 ll=173342 dtm=200 tn=173138 lm=172938
18.04.2022 17:43:39 : gg nn=200 ll=172336 dtm=200 tn=172138 lm=171938
18.04.2022 17:43:38 : gg nn=200 ll=171334 dtm=200 tn=171138 lm=170938
18.04.2022 17:43:37 : gg nn=201 ll=170314 dtm=200 tn=170138 lm=169938
18.04.2022 17:43:36 : gg nn=200 ll=169310 dtm=200 tn=169138 lm=168938
18.04.2022 17:43:35 : gg nn=200 ll=168308 dtm=200 tn=168138 lm=167938
18.04.2022 17:43:34 : gg nn=203 ll=167107 dtm=200 tn=166938 lm=166738
18.04.2022 17:43:33 : gg nn=201 ll=166102 dtm=336 tn=165938 lm=165602
18.04.2022 17:43:33 : jitt 165871 165000 165602
18.04.2022 17:43:32 : gg nn=221 ll=165499 dtm=200 tn=165202 lm=165002
18.04.2022 17:43:31 : gg nn=201 ll=164477 dtm=200 tn=164202 lm=164002
18.04.2022 17:43:30 : gg nn=201 ll=163474 dtm=200 tn=163202 lm=163002
18.04.2022 17:43:29 : gg nn=200 ll=162470 dtm=200 tn=162202 lm=162002
18.04.2022 17:43:28 : gg nn=200 ll=161467 dtm=200 tn=161202 lm=161002
18.04.2022 17:43:27 : gg nn=221 ll=160466 dtm=200 tn=160202 lm=160002
18.04.2022 17:43:27 : gg nn=200 ll=160044 dtm=200 tn=159802 lm=159602
18.04.2022 17:43:26 : gg nn=200 ll=159041 dtm=200 tn=158802 lm=158602
18.04.2022 17:43:25 : gg nn=201 ll=158037 dtm=200 tn=157802 lm=157602
18.04.2022 17:43:24 : gg nn=200 ll=157035 dtm=200 tn=156802 lm=156602
18.04.2022 17:43:23 : gg nn=200 ll=156031 dtm=200 tn=155802 lm=155602
18.04.2022 17:43:22 : gg nn=200 ll=155025 dtm=200 tn=154802 lm=154602
18.04.2022 17:43:21 : gg nn=220 ll=154022 dtm=200 tn=153802 lm=153602
18.04.2022 17:43:20 : gg nn=200 ll=153000 dtm=200 tn=152802 lm=152602
18.04.2022 17:43:19 : gg nn=202 ll=151997 dtm=200 tn=151802 lm=151602
18.04.2022 17:43:18 : gg nn=200 ll=150992 dtm=200 tn=150802 lm=150602
18.04.2022 17:43:17 : gg nn=202 ll=149988 dtm=200 tn=149802 lm=149602
18.04.2022 17:43:16 : gg nn=201 ll=148984 dtm=200 tn=148802 lm=148602
18.04.2022 17:43:15 : gg nn=202 ll=147981 dtm=200 tn=147802 lm=147602
18.04.2022 17:43:14 : gg nn=200 ll=146956 dtm=200 tn=146802 lm=146602
18.04.2022 17:43:13 : gg nn=200 ll=145953 dtm=333 tn=145802 lm=145469
18.04.2022 17:43:13 : jitt 145736 145000 145469
18.04.2022 17:43:13 : gg nn=200 ll=145552 dtm=200 tn=145269 lm=145069
18.04.2022 17:43:12 : gg nn=200 ll=144549 dtm=200 tn=144269 lm=144069
18.04.2022 17:43:11 : gg nn=200 ll=143548 dtm=200 tn=143269 lm=143069
18.04.2022 17:43:10 : gg nn=200 ll=142542 dtm=200 tn=142269 lm=142069
18.04.2022 17:43:09 : gg nn=220 ll=141538 dtm=200 tn=141269 lm=141069
18.04.2022 17:43:08 : gg nn=201 ll=140514 dtm=200 tn=140269 lm=140069
18.04.2022 17:43:07 : gg nn=200 ll=139507 dtm=200 tn=139269 lm=139069
18.04.2022 17:43:06 : gg nn=200 ll=138503 dtm=200 tn=138269 lm=138069
18.04.2022 17:43:05 : gg nn=201 ll=137499 dtm=200 tn=137269 lm=137069
18.04.2022 17:43:04 : gg nn=200 ll=136496 dtm=200 tn=136269 lm=136069
18.04.2022 17:43:03 : gg nn=201 ll=135494 dtm=200 tn=135269 lm=135069
18.04.2022 17:43:02 : gg nn=201 ll=134490 dtm=200 tn=134269 lm=134069
18.04.2022 17:43:01 : gg nn=201 ll=133468 dtm=200 tn=133269 lm=133069
18.04.2022 17:43:00 : gg nn=201 ll=132462 dtm=200 tn=132269 lm=132069
18.04.2022 17:42:59 : gg nn=202 ll=131456 dtm=200 tn=131269 lm=131069
18.04.2022 17:42:58 : gg nn=200 ll=130449 dtm=200 tn=130269 lm=130069
18.04.2022 17:42:57 : gg nn=200 ll=129446 dtm=200 tn=129269 lm=129069
18.04.2022 17:42:56 : gg nn=200 ll=128445 dtm=200 tn=128269 lm=128069
18.04.2022 17:42:55 : gg nn=200 ll=127442 dtm=200 tn=127269 lm=127069
18.04.2022 17:42:54 : gg nn=221 ll=126438 dtm=200 tn=126269 lm=126069
18.04.2022 17:42:53 : gg nn=201 ll=125816 dtm=335 tn=125669 lm=125334
18.04.2022 17:42:53 : jitt 125601 125000 125334
18.04.2022 17:42:52 : gg nn=200 ll=125215 dtm=200 tn=124934 lm=124734
18.04.2022 17:42:51 : gg nn=200 ll=124211 dtm=200 tn=123934 lm=123734
18.04.2022 17:42:50 : gg nn=201 ll=123208 dtm=200 tn=122934 lm=122734
18.04.2022 17:42:49 : gg nn=200 ll=122203 dtm=200 tn=121934 lm=121734
18.04.2022 17:42:48 : gg nn=202 ll=121199 dtm=200 tn=120934 lm=120734
18.04.2022 17:42:47 : gg nn=200 ll=120193 dtm=200 tn=119934 lm=119734
18.04.2022 17:42:46 : gg nn=221 ll=119190 dtm=200 tn=118934 lm=118734
18.04.2022 17:42:46 : gg nn=200 ll=118769 dtm=200 tn=118534 lm=118334
18.04.2022 17:42:45 : gg nn=200 ll=117765 dtm=200 tn=117534 lm=117334
18.04.2022 17:42:44 : gg nn=201 ll=116759 dtm=200 tn=116534 lm=116334
18.04.2022 17:42:43 : gg nn=200 ll=115754 dtm=200 tn=115534 lm=115334
```

WickedShell, 2022-04-19:
Sorry, I e-mailed something back about this before, but github lost it. We need to handle to handle this better, but last I investigated this is actually due to a bad clock on the flight controller board. The cheaper boards sometimes omit tuning capacitors, which leads to badly out of spec clock drift, which in turn leads to an error in the jitter calculation. If you have a long log of just GPS messages you can calculate the clock drift. A 30 minute to an hour long log is plenty long enough.

Sekilsgs2, 2022-04-20:
> Sorry, I e-mailed something back about this before, but github lost it. We need to handle to handle this better, but last I investigated this is actually due to a bad clock on the flight controller [...]

Yes i'm understand that some board have time drift between real (from gps) and local - but this should not affect delta calculation - because delta this is the difference between GPS frames but not difference beetwen GPS and local clocks - and GPA.delta spikes (~340ms) - not true - this only bug in calculations. 
you need to take into account time drift when calculating lags...

Vkd4U2NtVkZOVg, 2022-05-11:
Those board often use murata CSTNE series resonators with built-in load 
capacitors , so It dosen't require external capacitors, It's not a design issue

Sekilsgs2, 2022-05-11:
> Those board often use murata CSTNE series resonators with built-in load capacitors , so It dosen't require external capacitors, It's not a design issue

Wow great comment - this is another stone in the direction of the developers and that the current implementation has errors and is not suitable and maybe not related to xtal...

Thanks!

Vkd4U2NtVkZOVg, 2022-05-11:
> This moment is not correctly taken into account in the current implementation of the calculation -
> delta_time_ms - if i'm using gps_rate_ms == 200ms i'm have gpa.delta spikes (~340ms) every 20 [...]

Although It's not a hardware design issue, but I think It's maybe related to the resonator itself... I study the CSTNE's [datasheet](https://www.murata.com/en-us/api/pdfdownloadapi?cate=cgsubResonators&partno=CSTNE8M00G55A000R0) It's a 8MHz and It's the most common frequency used on those STM32F405 based fly controller. according to the datasheet
The Frequency Tolerance is about +/-0.50% max (8 * 0.005 = 400KHz), and I have a another 16MHz CRYSTAL's [datasheet](https://www.mouser.com/datasheet/2/137/FA_128_en-931416.pdf)  that I used on my fmuv5 hardware, and the Frequency tolerance is 30ppm(for clock application) which is 48Hz......

Sekilsgs2, 2022-05-11:
> It's maybe related to the resonator itself

:) and yet the resonator...

But again - the current code does not take this into account at all .. so you can reject half of all existing flight controllers, and some just stop using Ardu because of such childhood illnesses.. 
and the inaction of developers who do not see point-blank that this moment can be easily corrected.. imho..

Vkd4U2NtVkZOVg, 2022-05-11:
well...of course, the code need to handle this situation better, but I used this resonator a lot on my DIY F405 FC, never see Unhealthy GPS signal warning popped up. I guess this is not that common issue. and also It is related to the manufacturer, the Frequency Tolerance is +/-0.50% max, and the Frequency Shift by Temperature is +/-0.20% max. so the solder temperature also affect the frequency drift.

Sekilsgs2, 2022-05-11:
> never see Unhealthy GPS signal warning popped up

Strange - i'm have two fc and two have this problems...
But again :) - from gps, we only need location data - which we receive regularly and without failures, but ardu does not correctly respond to what needs to be skipped but does not create problems in any way. It's all solvable...

WickedShell, 2022-05-11:
> Those board often use murata CSTNE series resonators with built-in load capacitors , so It dosen't require external capacitors, It's not a design issue

I can't speak to all of them, but I've gotten logs from multiple cheaper flight controllers that have clock drift at 3000 PPM or more. If you want to send me an hour long DF log with GPS logging we can calculate the observed clock drift on the board you are using and rule that out if you would like. (I'm also always happy to see what more boards drift looks like). I only need the GPS/GPA messages in the DF log, everything else can be turned off to take less space if you want.

Vkd4U2NtVkZOVg, 2022-05-12:
> have clock drift at 3000 PPM or more

It's a resonable value according to the datasheet, the maximum drift can be 5000ppm.

WickedShell, 2022-10-11:
This is an error from the jitter correction code. But the underlying error
is actually that the board you are using has a very bad clock drift, rather
then a direct error in the jitter math.

Sekilsgs2, 2022-10-11:
this does not affect flights in any way in my case - and the error message and strange code are misleading - which requires reworking or the ability to turn it all off.