ArduPilot/ardupilot #20567: Matek H743-WING vsw relay not working

pedrofdmp-devops, 2022-04-18:
## Bug report

**Issue details**

Matek H743-WING vsw relay not working, configuration (power to the vtx is not cut):
RELAY_PIN,81
RELAY_PIN2,82
RC6_OPTION,34
RC7_OPTION,28

Tested with arduplane  4.2.0beta5 and 4.1.7_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
flying wing

**Hardware type**
Matek H743-WING

**Logs**
[(https://www.youtube.com/watch?v=MQnqAPaW0mM)](https://www.youtube.com/watch?v=MQnqAPaW0mM)

rmackay9, 2022-04-19:
@pedrofdmp-devops,

Could you try setting SERVOn_FUNCTION = -1 for the channel used for the relay?

Hwurzburg, 2022-04-19:
this is not a pwm output issue...these are dedicated GPIOS....not servo function....setup is correct....sounds like a board issue...the VSW output transistor can fail if overstressed (overcurrent by accidentally shorting is common) either open or shorted and not switch

pedrofdmp-devops, 2022-04-20:
@Hwurzburg , there was no short afaik and the camera switch is working fine.

I posted on rcgroups to see if anyone with a similar board can test this: https://www.rcgroups.com/forums/showthread.php?3692769-Matek-H743-Wing-Flight-Controller-w-Ardupilot-%28ChiBiOS%29/page38#post49052413

In the mean time I will try to find the transistor on the board and test it.

pedrofdmp-devops, 2022-04-23:
Decided to try a F765-WING flight controller and got the same result, the camera switch works, but the vsw switch does not:
[image]
[https://www.youtube.com/watch?v=I--o1Og7pXk](url)

Hwurzburg, 2022-12-13:
I think the solder jumper on the back has not be set for which supply is needed....this is the common reason for this failure...closing this....reopen if you think this was not the issue

closed 2022-12-13 by Hwurzburg