ArduPilot/ardupilot #20582: Copter: slow parameter download after _ENABLE set to 1

rmackay9, 2022-04-19:
With Copter-4.3.0-DEV (and 42.0-rc1) on a CubeOrange, the parameter download (via MAVFTP) can take about 30seconds if some _ENABLE parameters are set.  In particular I found I was able to recreate the issue on a CubeOrange which had been reset to factory settings BTN_ENABLE = 1 OR CAN_P1_DRIVER = 1.

Enabling these parameters (and rebooting the autopilot) would make the parameter download slow.
Setting these parameters back to 0 (and rebooting) would make the parameter downloads fast again

Reverting the autopilot to Copter-4.1.5 would also resolve the problem.  Re-installing Copter-4.3.0-DEV would immediately cause the problem to re-appear.

FYI @tridge @meee1

It may not be helpful but this is the message the MP displays during this 30 seconds.
[image]

Sekilsgs2, 2022-04-20:
I'm have slow with btn not enabled

rmackay9, 2022-04-25:
Here's a parameter file for a CubeOrange that reproduces the problem https://www.dropbox.com/s/1bmj120e5d4ukxa/randy-param-slow-cube-orange-25Apr2022.param?dl=0

tridge, 2022-04-25:
it appears to be a MissionPlanner issue, or at least when it is in this state mavproxy isn't slow

tridge, 2022-04-25:
ping @meee1

meee1, 2022-04-25:
ok so.
when MP send a kCmdResetSessions
i get a responce thats not valid from AP
i send
opcode:2	size:0	req_opcode:0	burst_complete:0	padding:0	offset:0
and i get a responce of
opcode:129	size:1	req_opcode:15	burst_complete:0	padding:0	offset:0

where req_opcode is the last command used before this one.
needs to be fixed
here
https://github.com/ArduPilot/ardupilot/blob/0e18bd450d07e7bc3805b0907e15791207f5f5a7/libraries/GCS_MAVLink/GCS_FTP.cpp#L218
and
https://github.com/ArduPilot/ardupilot/blob/0e18bd450d07e7bc3805b0907e15791207f5f5a7/libraries/GCS_MAVLink/GCS_FTP.cpp#L248

Mission planner always matchs responses with the request.

rmackay9, 2022-04-26:
Thanks to @meee1 this issue no longer happens with the latest MP.
[image]

I'm not actually sure if the issue was a bug in MP or a bug in AP that MP now handles but at least our users won't be inconvenienced any more.  Txs!

closed 2022-04-26 by rmackay9

robustini, 2022-08-08:
In my opinion the problem has not yet been solved.
Classic chinese telemetry 433 MHz with SiK firmware V2.2.
With the MavFTP enabled the download of the parameters is infinite and never end, if I disable it it is fast as a rocket.

Mith MavFTP:

[image]

Without:

[image]