ArduPilot/ardupilot #20583: Neopixel broken in transition from AP-Periph v1.3 to v1.4dev

MallikarjunSE, 2022-04-19:
## Bug report

**Issue details**
AP-Periph LEDs behave wrong even when node is being sent with right LED colour data. 

I want to report this so that bug can be found between the dates of AP-Periph v1.3 stable and date of report. 
Found this issue from 22/03/2022 master build. Even the start sequence is different from normal. 

**Version**
_What version was the issue encountered with_

**Platform**
AP-Periph

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
Sierra-F412, Sierra-F405

**Links**
https://discord.com/channels/674039678562861068/777699265069187102/964390939365888030
https://discord.com/channels/674039678562861068/777699265069187102/964391412852482098
LEDs start up sequence:
https://discord.com/channels/674039678562861068/777699265069187102/955748588326092840

andyp1per, 2022-04-19:
@MallikarjunSE can you see if https://github.com/ArduPilot/ardupilot/pull/20432 fixes?

MallikarjunSE, 2022-04-19:
> @MallikarjunSE can you see if https://github.com/ArduPilot/ardupilot/pull/20432 fixes?

Will try it now!

MallikarjunSE, 2022-04-19:
Hey @andyp1per still same. 
https://photos.app.goo.gl/6xBUrd7ihJEYxKxA7

tridge, 2022-05-15:
@MallikarjunSE we had a bug in master that defaulted NTF_LED_OVERRIDE to the wrong value on peripherals. The result was the LED stayed yellow. The bug was never in a stable peripheral release
the fix was here:
https://github.com/ArduPilot/ardupilot/commit/12f348398ad92dcfbb66cbfe0591aeaa2c268a8e

MallikarjunSE, 2022-05-16:
@tridge got it. Thanks!

closed 2022-05-16 by MallikarjunSE