ArduPilot/ardupilot #20585: AP_VisualOdom, AP_EKF3: external position estimates treated as offsets from internal moving origin

rmackay9, 2022-04-20:
A user reported that positions provided via VISION_POSITION_ESTIMATE mavlink messages are being treated as if they are offsets from the internal moving origin (added as part of the "Extreme ArduPilot" project) instead of as offset from the non-moving (external?) EKF origin.

In some ways, it makes sense to use the internal origin so that external vision systems can be used at long distances from the EKF origin but on the other hand, it certainly causes problems at the moment the internal origin is moved because the external system has no way to know where the internal origin is.

The short-term fix (at least) is to treat the vision-position-estimate messages as offset from the non-moving (external?) EKF origin.

xianglunkai, 2022-04-20:
Should motion compensation of internal origin be considered, including angular velocity and linear velocity?

chobitsfan, 2022-04-20:
I think in current code,  ext nav data is already treated as offset from their own origin.  When using ext nav, EKF is using ext nav origin as its origin.
In `NavEKF3_core::ResetPosition`
https://github.com/ArduPilot/ardupilot/blob/05bda89516bfdc870adb22f96cbd7367e506eb76/libraries/AP_NavEKF3/AP_NavEKF3_PosVelFusion.cpp#L117-L118
compare to 
https://github.com/ArduPilot/ardupilot/blob/05bda89516bfdc870adb22f96cbd7367e506eb76/libraries/AP_NavEKF3/AP_NavEKF3_PosVelFusion.cpp#L100

rmackay9, 2022-04-26:
We discussed this on the dev call and I will do a simple test in SITL where we fly the vehicle in Circle mode and if it's broken it won't fly correctly.

rmackay9, 2022-04-26:
I've tested the Vicon in SITL in Circle mode and it seems fine so I think we can assume that we are handling the internal vs external EKF origin correctly.
[image]

closed 2022-04-26 by rmackay9