ArduPilot/ardupilot #20587: Diatone Mamba F405 MK4 Support

Sekilsgs2, 2022-04-20:
Please add Diatone Mamba F405 MK4 to supported boards.

BF target - 

https://github.com/betaflight/unified-targets/blob/master/configs/default/DIAT-MAMBAH743.config

Allentgt, 2022-06-03:
Any update on support for this board? Is this being looked into?

andyp1per, 2022-06-03:
I have the mini so will port that when I have done the H743 which is proving more difficult that I thought

Allentgt, 2022-06-04:
That's great. I was just asking because I'm not able to get my hands on the matek h743 and this seemed like a good alternative.

closed 2022-06-08 by peterbarker

Sekilsgs2, 2022-06-08:
Wow super thanks!

redbananamoon, 2023-07-31:
Please change the name of this FC in your documentation and firmware files. The name "Mamba H743 v4" is
likely to cause a lot of confusion, especially now that there is a "Mamba MK4 H743 V2". Also on this topic, is the newer "Mamba MK4 H743 V2" supported by the same firmware?

andyp1per, 2023-08-07:
Mk4 v2 is supported by the same firmware

PR #20902 (2022-06-08): Mamba H743v4 Support
Fixes #20587