ArduPilot/ardupilot #20592: why not function boundary.reset() is not called by all instances of AP_Proximity?

xianglunkai, 2022-04-21:
Although each face of the boundary has a timeout reset time of one second, I still think it should be reset for boundary when updating a new boundary in any backends of AP_Proximity.

rishabsingh3003, 2022-04-21:
@xianglunkai we reset each individual face in all our drivers by calling boundary.reset_face(). We expect the drivers to be updating each face individually, and constantly. The timeout is merely a backup (failsafe) so we don't leave any face unattended, and we don't really expect it to ever be used normally.

boundary.reset() would wipe the entire boundary which isn't the behaviour we are looking for. Many sensors constantly stream live data over several iterations. Calling boundary.reset() again and again on each iteration for such drivers would mean loosing out on a chunk of obstacles on the boundary.

xianglunkai, 2022-04-21:
@rishabsingh3003 Txs,you are right!
I just saw that when the vehicle turned quickly, the obstacles maybe in the boundary sector dragged for a short time.

closed 2022-04-21 by xianglunkai