ArduPilot/ardupilot #20597: #20379 breaks DSHOT

shellixyz, 2022-04-21:
e0c0f4ea3410656b75748287cf59a864d8e0518d from #20379 breaks DSHOT at least on F405 and using BLHeli_S

How to reproduce:
- Flash a quad with master (with a F405 FC and BLHeli_S ESCs to be sure it doesn't work)
- Configure the outputs as DSHOT
- Watch and listen to the motors as the ESC doesn't properly initialize (tones and motors randomly running or not)

andyp1per, 2022-04-21:
It also makes BlueJay work where they didn't before, so seems like we are damned if we do and damned if we don't. Maybe we need a configuration option around this. What BLHeli_S ESCs are you using?

shellixyz, 2022-04-21:
I guess an option would be the solution. I'm going to test with a BLH32 ESC to see if that makes a difference. I'm using this ESC on this quad:

https://www.racerstar.com/racerstar-rs20ax4-20a-4-in-1-blheli_s-opto-esc-2-4s-support-dshot150-dshot300-for-rc-fpv-racing-drone-p-64.html

shellixyz, 2022-04-21:
Seems to be working fine with BLH32

andyp1per, 2022-04-22:
Are you able to try bluejay on that ESC to see if its any different?

Since you are able to build the firmware I would be interested in knowing a few things:

- Are all dshot rates broken or just some?
- What is the output of @SYS/timers.txt on your board?
- if you pass false to the last argument of calculate_bitrate_prescaler() does it help?
- Are there any adjacent values of DSHOT_BIT_0_TICKS and DSHOT_BIT_1_TICKS that work?

peterbarker, 2024-07-31:
Ping @shellixyz - @andyp1per asked some questions a while back which weren't answered.

Still having problems?

shellixyz, 2024-07-31:
It is probably fine now I think the ESC type setting when it was introduced fixed it. I don't exactly recall now to be honest and I haven't used a recent version of Ardupilot in a while. I cannot test this for at least a few months either. I'm closing this issue

closed 2024-07-31 by shellixyz