ArduPilot/ardupilot #20604: Airspeed  I2C MS4525D0

Magnorio, 2022-04-22:
## Bug report

**Issue details**
Airspeed sensor I2C MS4535D0 is working on FW4.1.7 but after upgrading to 4.2 isnt working anymore.
AP sends message pre arm error: "airspeed health"

**Version**
4.2 Beta5

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
glider

**Hardware type**
Mateksys H743 Wing

**Logs**

nikandriya, 2022-05-27:
try erasing the chip, and reflashing again. Very likely it is ardupilot parameter corruption problem.

hendjoshsr71, 2022-05-27:
Before you go about erasing anything, please send a log file.

To collect this set LOG_DISARMED 1

I want to be sure there is no parameter conversion issue.

DenisMaliy, 2026-09-08:
Same symptom on Matek H743-WING V3 + Matek ASPD-4525 (MS4525DO) — root-caused with a scope: I²C edge rate / crosstalk, not the driver

I hit what looks like exactly this issue and spent some time isolating it, so posting the findings in case they help.

Setup

Matek H743-WING V3, ArduPlane 4.7.1 stable (also reproduced on a 4.8.0-dev build, 214e1f94)
Matek ASPD-4525 (MS4525DO, I²C addr 0x28) on the I2C2 pads (DA2/CL2, ARSPD_BUS 0), stock 20 cm Matek cable
ARSPD_TYPE 1, ARSPD_OPTIONS 0, ARSPD_USE 0/1 (no difference)

Symptom
Sensor is detected (MS4525[0]: Found bus 0 addr 0x28, ARSPD_DEVID 141313), calibrates, but the AIRSPEED MAVLink message alternates between good frames (flags=2, raw_press ≈ 5 Pa, temperature 28.29 °C) and corrupt ones (flags=1, raw_press ≈ 357 Pa, temperature 32767). Both pressure and temperature go bad together, i.e. the whole 4-byte frame is corrupted, not a single reading. Majority of frames were bad. With ARSPD_USE 1 the DCM horizon visibly jumps on every good↔bad transition.

What did NOT change anything

ARSPD_OPTIONS 0, USB vs battery power, removing pitot tubes, twisting wires
Moving the sensor to I2C1 (ARSPD_BUS 1) — identical behaviour
4.7.1 stable vs 4.8.0-dev — identical
An HMC5883L compass on the same bus and same wires is rock solid (RAW_IMU xmag/ymag noise ±3 mGauss) — so the bus itself works for other devices
The same sensor + same cable on a SpeedyBee F405 V4 with ArduPlane 4.7.1 is perfectly clean — so the sensor is fine

Scope measurements (Siglent SDS1204X-E, 200 MHz, probed at the sensor pads)

Bus clock ≈ 100 kHz, rise times 218 ns (SCL) / 225 ns (SDA) — fine
Fall time SCL: 2.27 ns, fall time SDA: 1.06 ns — the H7 drives the open-drain lines low extremely fast (the I²C fast-mode spec sets a minimum fall time of ~20·(VDD/5.5 V) ns ≈ 12 ns precisely to limit this)
Every edge on one line induces a ~1.1 V glitch on the other line (3.3 V → ~2.2 V) with 40–60 ns of ringing on the 20 cm cable. That lands right at the MS4525 input threshold (0.7·VDD = 2.31 V). The HMC5883L input apparently tolerates it; the MS4525 does not, and occasionally sees a false START/STOP or an extra clock → shifted/garbage frame.

Fix (hardware)

51 Ω series in SDA and SCL at the FC end alone did almost nothing (cable capacitance is only ~20 pF, so RC ≈ 1 ns).
51 Ω series at the FC end plus 100 pF to GND on SDA and SCL at the sensor end: fall times become 13.5 ns (SCL) / ~19 ns (SDA), crosstalk drops to 0.15–0.3 V, rise time 700 ns (still within the 1 µs limit at 100 kHz). After that: 10 min at 10 Hz → zero corrupted frames, one missed read (temperature 32767 with pressure unchanged) in ~6000 samples.

Suggestion
This looks like a GPIO output-speed issue on H7 targets rather than an AP_Airspeed driver bug. ArduPilot hwdef supports per-pin SPEED_LOW / SPEED_MEDIUM; if the I²C pins on H7 boards are currently at the default (fast) slew rate, lowering it in hwdef for the I²C pins might fix this in firmware for everyone with "weak-input" I²C devices on cables, without external RC.

Scope captures before/after attached.

AIRSPEED message, good frame vs corrupted frame (pressure and temperature go bad together)
[image]

[image]

SCL falling edge at the sensor, stock wiring: 2.27 ns, ~1 V glitch coupled into SDA (20 ns/div)
[image]

Same edge with 51 Ω series (FC end) + 100 pF to GND (sensor end): 13.5 ns, crosstalk ~0.2 V
[image]