ArduPilot/ardupilot #20608: Copter: ATC_SLEW_YAW should not affect Loiter

rmackay9, 2022-04-23:
There are three similar parameters which control the rate that a copter can rotate in yaw:

- [PILOT_Y_RATE](https://ardupilot.org/copter/docs/parameters.html#pilot-y-rate-pilot-controlled-yaw-rate)
- [ATC_RATE_Y_MAX ](https://ardupilot.org/copter/docs/parameters.html#atc-rate-y-max-angular-velocity-max-for-yaw)
- [ATC_SLEW_YAW](https://ardupilot.org/copter/docs/parameters.html#atc-slew-yaw-yaw-target-slew-rate)

[ATC_SLEW_YAW](https://ardupilot.org/copter/docs/parameters.html#atc-slew-yaw-yaw-target-slew-rate) should not affect the rate of rotation of Loiter mode (even though the parameter descriptions says that it does)

FYI @lthall

IamPete1, 2022-04-23:
We did have the opposite issue to this....  https://github.com/ArduPilot/ardupilot/issues/12600

MichelleRos, 2022-04-26:
I noticed this on update to rc1 as well (from a few weeks old master). That default rotation rate in Loiter was essentially unusable for me. I had to switch to Alt_hold regain control (luckily this was on a near-windless day).

rmackay9, 2022-05-18:
This has been resolved in -rc4 so closing.

closed 2022-05-18 by rmackay9

shubham-shahh, 2023-07-18:
what's the difference between ATC_SLEW_YAW and ATC_RATE_Y_MAX