ArduPilot/ardupilot #20609: AP_Button: internal error triggered if no PWM input

rmackay9, 2022-04-23:
I can somewhat easily recreate an internal error using AP_Button if setup to use PWM input.

PreArm: Internal errors 0x2000000 l:554 gpio_isr

I've passed a log to @peterbarker in the hopes that he can help track down the issue.

peterbarker, 2022-04-23:
Is the PWM input zero or floating?

What you're seeing is the ISR flood counter tripping - it's an explicit check for receiving an interrupt way, way too often.

We might be able to get this if we're emitting data on the same pad - any d-shot involved or something?

rmackay9, 2022-05-09:
This only seems to happen when my servo tester is powered off but still connected to the autopilot's PWM input pins.  If the servo tester is disconnected or powered on then it's fine.  I've also checked that the pull-up is working so I think what we're seeing is a fight between the autopilot's pull-up and the servo tester when powered off.

Anywya, I think this is a hardware specific issue and is not a real problem so closing. txs!

closed 2022-05-09 by rmackay9