ArduPilot/ardupilot #20611: The f4v3s elrs 915Mhz remote control signal has a large delay, and there is no such problem when using inav.

wanglie10, 2022-04-23:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**

_Please describe the problem_

**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

IamPete1, 2022-04-23:
What version of ArduPilot? What hardware? How is it connected?

Sekilsgs2, 2022-04-24:
I'm have no problems with AP and my elrs 915mhz. 
Maybe problem in something else - share your paramsm or binary log

rmackay9, 2022-04-25:
This should really go into [the support forums](https://discuss.ardupilot.org/c/arducopter/6) so I will close this issue. Sorry to be strict but if we're not then our to-do list turns into another support forum and we lose track of confirmed bugs and enhancment requests.

closed 2022-04-25 by rmackay9