ArduPilot/ardupilot #20628: Ardupilot Loiter Brake

lixiaowei123, 2022-04-26:
Ardupilot Loiter Brake 
The braking effect is not good. Can you optimize or add position control

rmackay9, 2022-04-26:
Hi @lixiaowei123,

Txs for the report.  Things can always be improved of course but I think we will need to see an onboard log and will need more details on what is "not good".

lixiaowei123, 2022-04-26:
After braking, the plane is leveled, but it will drift a little. Normally, the plane should stop directly after braking and flattening

lixiaowei123, 2022-04-26:
@rmackay9 hi
roll_out = (18000/M_PI) * atanf(cosf(pitch_out*(M_PI/18000))*tanf(roll_out*(M_PI/18000))); Why is this handled like this

lthall, 2022-04-26:
Hi @lixiaowei123 

Randy requested a log and a description of what is "not good". You have not provided a log as requested.

This is not a support forum. Questions like the one above should be directed to the discuss forum.

Loiter does stop after breaking. By this I mean the commanded velocity is zero. If there is any drift then it is most likely a result of your tuning or problems with your sensor setup.

As you have not followed the instructions on this issue or provided the requested information I am closing this issue.  These instructions are to help avoid waisting multiple developers time.

For reference these are the instructions you deleted before writing your very brief complaint:

**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**

_Please describe the problem_

**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_