ArduPilot/ardupilot #20637: can't switch on RELAY0 after arming when motor type is set to BrushedWithRelay

jazzl0ver, 2022-04-27:
## Bug report

**Issue details**

After firmware upload, the parameters were reset to their defaults and then applied these and rebooted:
```
ARMING_CHECK,0
BRD_PWM_COUNT,2
BRD_SAFETYENABLE,0
MOT_PWM_TYPE,3
RELAY_PIN,52
RELAY_PIN2,53
RELAY_PIN3,54
RELAY_PIN4,55
SERVO11_FUNCTION,-1
SERVO12_FUNCTION,-1
SERVO13_FUNCTION,-1
SERVO14_FUNCTION,-1
```

The testing procedure:
A voltmeter is connected to the control pins of AUX3 and AUX4. Before arming, a voltage has been measured between the pins when RELAY0 is set to HIGH and RELAY1 is set to LOW and vice versa using Mission Planner's Servo/Relay tab. In both cases, the voltage raised to +-~3.3V.
After arming the same measurement has been conducted with the following results:
RELAY0:HIGH, RELAY1:LOW - voltage starting to grow but then quickly goes back to 0V
RELAY0:LOW, RELAY1:HIGH - voltage is 3.3V

Voltage between to the control pins of AUX5 and AUX6 is +-3.3V as expected regardless of the arming state.

Changing motor type from BrushedWithRelay to Normal (and rebooting) gets rid of this issue.

**Version**
4.3.0-dev
4.2.0-rc1
4.1.5-stable
3.5.2-stable

**Platform**
[X] Rover

**Airframe type**
N/A

**Hardware type**
Pixhack 2.8.4 (a Pixhawk clone)

**Logs**
N/A

IamPete1, 2022-04-27:
Brushed with relay uses relay 0 through 4 depending on how many motors have setup. When configured for this this you cannot set the relays manually when the motors are active.

jazzl0ver, 2022-04-27:
@IamPete1 thanks for quick reply! Is this behavior documented somewhere (I checked Brushed Motors and Relay wiki pages and found nothing regarding this)?

I was able to shift the relays and work with the pins 3-6 (AUX3-AUX6):
```
RELAY_PIN,50
RELAY_PIN2,51
...
RELAY_PIN6,55
SERVO9_FUNCTION,-1
...
SERVO14_FUNCTION,-1
```
And now AUX3 pin works as expected. Thanks!

In my setup I use two servo outputs: PWM1 is for steering, PWM2 is for throttle. Can you please explain why I had problems only with RELAY0?

jazzl0ver, 2022-04-28:
No docs on that, right?

closed 2022-04-28 by IamPete1

jazzl0ver, 2022-05-30:
Finally completed a short guide on how to make things work:
https://github.com/jazzl0ver/ardupilot-rover-l298n

Thanks again for your help!