ArduPilot/ardupilot #20647: Helicopter frame in the help message of sim_vehicle.py

tatsuy, 2022-04-27:
I found Helicopter in the options of sim_vehicle.py, but Helicopter does not exist as a vehicle type.
It's added for autotest. #17701

```
$ sim_vehicle.py -h
Usage: sim_vehicle.py

Options:
  -h, --help            show this help message and exit
  -v VEHICLE, --vehicle=VEHICLE
                        vehicle type (Rover|ArduSub|Blimp|ArduPlane|ArduCopter
                        |AntennaTracker|Helicopter)                               <--------------
  -f FRAME, --frame=FRAME
                        set vehicle frame type
                        Rover: airsim-rover|balancebot|calibration|gazebo-
                            rover|motorboat|rover|rover-skid|rover-
                            vectored|sailboat|sailboat-motor
                        ArduSub: gazebo-bluerov2|vectored|vectored_6dof
                        Blimp: Blimp
                        ArduPlane: CRRCSim|calibration|firefly|gazebo-
                            zephyr|jsbsim|last_letter|plane|plane-dspoilers
                            |plane-elevon|plane-jet|plane-soaring|plane-
                            tailsitter|plane-vtail|quadplane|quadplane-cl84
                            |quadplane-copter_tailsitter|quadplane-tilthvec
                            |quadplane-tilttri|quadplane-tilttrivec|quadplane-
                            tri|scrimmage-plane
                        ArduCopter: +|Callisto|IrisRos|X|airsim-
                            copter|bfx|calibration|coaxcopter|cwx|deca|deca-
                            cwx|djix|dodeca-hexa|gazebo-iris|heli|heli-
                            blade360|heli-dual|hexa|hexa-cwx|hexa-dji|octa
                            |octa-cwx|octa-dji|octa-quad|octa-quad-cwx|quad
                            |scrimmage-copter|singlecopter|tri|y6
                        AntennaTracker: tracker
                        Helicopter: heli|heli-blade360|heli-dual             <--------------
```

rmackay9, 2022-04-27:
FYI @peterbarker

Maarrk, 2026-09-25:
This is still present, and causes an error like this:
```sh
$ sim_vehicle.py -v Helicopter
vehicle directory (/home/marek/repos/ardupilot/Helicopter) does not exist
SIM_VEHICLE: Killing tasks
```

MichelleRos, 2026-09-25:
@Maarrk I don't know what parameter file you're using, but I just tested the standard one with `sim_vehicle.py -v ArduCopter -f heli` (after wiping the eeprom.bin) and it started up correctly and it took of in Loiter mode without issue.

Maarrk, 2026-10-05:
This seems to be related to RealFlight simulation. I was attempting:
```sh
sim_vehicle.py -v ArduCopter -f heli -f flightaxis:172.27.32.1 ...
```
But if I change the order of frame arguments, it works:
```sh
sim_vehicle.py -v ArduCopter -f flightaxis:172.27.32.1 -f heli ...
```

Comparing this to X-Plane, there is `-f xplane` for planes and `-f xplane-heli` for helicopters

khancyr, 2026-10-05:
> But if I change the order of frame arguments, it works:
> 
> sim_vehicle.py -v ArduCopter -f flightaxis:172.27.32.1 -f heli ...
> 
> Comparing this to X-Plane, there is `-f xplane` for planes and `-f xplane-heli` for helicopters

Woooops ... that is wrong, we shouldn't accept to pass multiple frame !

Maarrk, 2026-10-06:
The issue reported here was fixed in #34631, I think this issue can be closed.

---

> Woooops ... that is wrong, we shouldn't accept to pass multiple frame !

I feel like it would be better to continue that conversation in a separate issue, but I'd need @khancyr to write their expectations of correct behaviour, thanks :)

closed 2026-10-06 by khancyr