ArduPilot/ardupilot #20650: Script: install-prereqs-ubuntu.sh  - fails on Ubuntu 22.04 (current release/LTE)  - isdir

AndKe, 2022-04-28:
## Bug report
```
Requirement already satisfied: pyserial>=3.0 in /usr/lib/python3/dist-packages (from MAVProxy) (3.5)
Requirement already satisfied: mccabe<0.7.0,>=0.6.0 in /home/andre/.local/lib/python3.10/site-packages (from flake8) (0.6.1)
Requirement already satisfied: pyflakes<2.5.0,>=2.4.0 in /home/andre/.local/lib/python3.10/site-packages (from flake8) (2.4.0)
Requirement already satisfied: pycodestyle<2.9.0,>=2.8.0 in /home/andre/.local/lib/python3.10/site-packages (from flake8) (2.8.0)
Requirement already satisfied: six in /usr/lib/python3/dist-packages (from geocoder) (1.16.0)
Requirement already satisfied: click in /usr/lib/python3/dist-packages (from geocoder) (8.0.3)
Requirement already satisfied: ratelim in /home/andre/.local/lib/python3.10/site-packages (from geocoder) (0.1.6)
Requirement already satisfied: requests in /usr/lib/python3/dist-packages (from geocoder) (2.25.1)
Requirement already satisfied: decorator in /usr/lib/python3/dist-packages (from ratelim->geocoder) (4.4.2)
Installing collected packages: flake8, MAVProxy, geocoder
  Attempting uninstall: MAVProxy
    Found existing installation: MAVProxy 1.8.48
ERROR: Exception:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/_internal/cli/base_command.py", line 165, in exc_logging_wrapper
    status = run_func(*args)
  File "/usr/lib/python3/dist-packages/pip/_internal/cli/req_command.py", line 205, in wrapper
    return func(self, options, args)
  File "/usr/lib/python3/dist-packages/pip/_internal/commands/install.py", line 405, in run
    installed = install_given_reqs(
  File "/usr/lib/python3/dist-packages/pip/_internal/req/__init__.py", line 68, in install_given_reqs
    uninstalled_pathset = requirement.uninstall(auto_confirm=True)
  File "/usr/lib/python3/dist-packages/pip/_internal/req/req_install.py", line 637, in uninstall
    uninstalled_pathset = UninstallPathSet.from_dist(dist)
  File "/usr/lib/python3/dist-packages/pip/_internal/req/req_uninstall.py", line 554, in from_dist
    for script in dist.iterdir("scripts"):
  File "/usr/lib/python3/dist-packages/pip/_internal/metadata/pkg_resources.py", line 156, in iterdir
    if not self._dist.isdir(name):
  File "/usr/lib/python3/dist-packages/pip/_vendor/pkg_resources/__init__.py", line 2816, in __getattr__
    return getattr(self._provider, attr)
AttributeError: 'EggMetadata' object has no attribute 'isdir'

```
**Issue details**
I have a local MAVproxy installed: (installed using ```setup.py build install --user```)
I suspect that is what would be removed.  (which would be nice if it was an option.)

**Version**
Current master

**Platform**
N/A

**Airframe type**
N/A

**Hardware type**
N/A 

**Logs**
please see above

alielbashir, 2022-07-03:
I also faced a similar issue, where most Python libraries installed from apt throw a ```package python-dev has no installation candidate```
error

rmackay9, 2022-07-04:
By the way, we have this PR https://github.com/ArduPilot/ardupilot/pull/21077 although I don't know if it will resolve the problem.

peterbarker, 2024-07-31:
I've tested the script recent on 22.04, so I believe this has been fixed.

Reopen if I'm wrong, please.

closed 2024-07-31 by peterbarker