ArduPilot/ardupilot #20651: Scripting not restarts correctly

mustafa-gokce, 2022-04-28:
## Bug report

So I create a scripts called `hello.lua`.
It contains:
`gcs:send_text(7, toString(0))`
Where `toString()` is not a predefined function, so what I get is:
```
AP: Lua: ./scripts/hello.lua:1: attempt to call a nil value (global 'toString')
```

When I correct it like below:
`gcs:send_text(7, tostring(0))`
And restart the scripting from MAVProxy with `scripting restart`, vehicle gets the command and:
`Got COMMAND_ACK: SCRIPTING: ACCEPTED`
Now the line works and prints but I still continue to get the messages from firmware like:
```
AP: 0
AP: Lua: ./scripts/hello.lua:1: attempt to call a nil value (global 'toString')
```

**Version**

AP: ArduCopter V4.3.0-dev (12d829e7)
AP: 4dc0a2c6bcf74e45b67ea2b22240cfe6

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
None

**Hardware type**
None

**Logs**
None

IamPete1, 2022-04-28:
Yep, should be fixed by  https://github.com/ArduPilot/ardupilot/pull/20549

IamPete1, 2022-05-18:
Closed by https://github.com/ArduPilot/ardupilot/pull/20549

closed 2022-05-18 by IamPete1